IHP-GmbH/IHP-Open-PDK #221: Separate KLayout DRC script and XML wrapper

mole99, 2024-10-15:
Currently, the DRC scripts are implemented using a `.lydrc` file, which is an XML wrapper around the DRC script: https://github.com/IHP-GmbH/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.tech/klayout/tech/drc

Instead, it would be great if the DRC script could be separated from the wrapper and simply included with `# %include`.
This would make it possible to run the DRC scripts from within a Ruby or Python script.

An example of how this works can be found in sky130. 
There is one [sky130A_mr.drc](https://github.com/efabless/sky130_klayout_pdk/blob/main/tech/sky130/drc/sky130A_mr.drc) DRC script that is included in several `.lydrc` files:

- [run_drc_beol.lydrc](https://github.com/efabless/sky130_klayout_pdk/blob/main/tech/sky130/macros/run_drc_beol.lydrc)
- [run_drc_feol.lydrc](https://github.com/efabless/sky130_klayout_pdk/blob/main/tech/sky130/macros/run_drc_feol.lydrc)
- [run_drc_full.lydrc](https://github.com/efabless/sky130_klayout_pdk/blob/main/tech/sky130/macros/run_drc_full.lydrc)

By setting the values of global variables, different DRC runsets can be enabled/disabled.

The same approach could be used to combine the `sg13g2_maximal` and `sg13g2_minimal` DRC scripts into a single file.

akrinke, 2024-10-23:
I agree that a single DRC script would be better because it avoids code duplication. However, due to the way we generate the scripts, some major changes are required. Therefore, it is not a priority for us at the moment.

In the end, it is not so different than for sky130: We have two `.lydrc` files, they have three. Besides, you can already run the `.lydrc` scripts from Ruby and Python:
```ruby
# ruby
RBA::Macro::new(path_to_lydrc).run()
```
Here is a full example:
```ruby
# DRC script configuration

unless $drc_script
  $drc_script = __dir__ + "/sg13g2_minimal.lydrc"
end

# END OF CONFIGURATION
######################

expanded_path = File.expand_path($drc_script)

start = Time.now
puts "Start: " + start.to_s

RBA::Macro::new(expanded_path).run()

finish = Time.now
puts "End:   " + finish.to_s

duration = finish - start
minutes = duration.to_i / 60
seconds = duration % 60

puts "Duration: %s%.0f s" % [minutes>0 ? "%d min " % [minutes] : "", seconds]
```

mole99, 2024-10-23:
Thanks for the reply! I've updated my comment above to note that it is possible to run a `.lydrc` file from Ruby/Python.

mole99, 2026-01-16:
Hi @sergeiandreyev,
Yes, this issue can be closed. The new DRC script is now properly separated from its XML macros.

closed 2026-01-16 by mole99