IHP-GmbH/IHP-Open-PDK #223: Rename the `ihp-sg13g2` folder to `sg13g2`

mole99, 2024-10-15:
AFAIK the name of the PDK is simply `sg13g2`? At least that's what it says in the [KLayout tech file](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/klayout/tech/sg13g2.lyt).

Could the `ihp-sg13g2` folder then be renamed to `sg13g2`? That way this folder can simply be copied to `PDK_ROOT` (where other PDKs may reside) and the `PDK` can be set to `sg13g2`.

This would require at least some changes, as for example xschem expects the PDK to be named `ihp-sg13g2`: https://github.com/IHP-GmbH/IHP-Open-PDK/blob/d3b389d94e4e4a394276c47d6be2ca15bc4a3bc2/ihp-sg13g2/libs.tech/xschem/xschemrc#L413

This might break the setup for some people, but better now than later as it is an uneccessary source of confusion. Thanks.

smunaut, 2024-10-15:
I was definitely asking myself that too and ATM I used `PDK=ihp-sh13g2` based on the name of the folder for the OpenLane2 support I've been working on. If that's not the name I should use, knowing quickly would definitely be good.

sergeiandreyev, 2024-10-15:
Hi @mole99, the full name of the PDK is `ihp-sg13g2`, `ihp` here is the same as `sky` in `sky130` and `gf` in `gf180mcu`
the `sg13g2` name is meant for internal use, and also for some backward compatibility w/ commercial PDK
maybe we should consider aligning this more carefully

mole99, 2024-10-15:
@sergeiandreyev Alright, thanks for the clarification! Yes, it would be great if this was consistent throughout the repository. I'll need to update the name in capiche too 😄

sergeiandreyev, 2024-10-18:
@mole99, could you show in details where the rename of `sg13g2` to `ihp-sg13g2` will make an impact? in some configuration settings?

mole99, 2024-10-22:
Hi Sergei, I'm not so sure anymore if it needs to be changed at all.
At first I was confused what is the full PDK name, as the other open source PDKs use the PDK base name + the variant (e.g. `sky130` + `A` or `gf180mcu` + `D`) and it is a bit different for `ihp-` + `sg13g2`. But now that this is cleared up, I don't think there's a need to change anything and `sg13g2` should just always be used internally. I can close this issue if you like.

sergeiandreyev, 2024-10-22:
Hi Leo, thank you for feedback!

closed 2024-10-22 by sergeiandreyev