IHP-GmbH/IHP-Open-PDK #226: Missing requirements: `psutil`

dnltz, 2024-10-18:
This PDK now has a dependency to `psutil`, which is not part of the standard Python library.

Packages shouldn't be installed directly into a system. I would suggest to introduce https://virtualenv.pypa.io/en/latest/ with a requirements file for this. Alternative, drop `psutil` and find a better alternative, for example `ps` inside a subprocess call.

```
ERROR: /home/daniel/work/aesc/i2c-gpio-expander/pdks/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/__init__.py:105: ModuleNotFoundError: No module named 'psutil'
  /home/daniel/work/aesc/i2c-gpio-expander/pdks/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/__init__.py:105
  /home/daniel/work/aesc/i2c-gpio-expander/pdks/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/__init__.py:151
  /home/daniel/work/aesc/i2c-gpio-expander/pdks/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/__init__.py:240
  /home/daniel/work/aesc/i2c-gpio-expander/pdks/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/pymacros/autorun.lym:8 (class ModuleNotFoundError)
ERROR: AttributeError: 'NoneType' object has no attribute 'layout'
  /home/daniel/work/aesc/i2c-gpio-expander/pdks/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout//tech/scripts/sealring.py:34
  /home/daniel/work/aesc/i2c-gpio-expander/pdks/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout//tech/scripts/sealring.py:70
```

ThomasZecha, 2024-10-19:
IMHO there is no reason to not use software besides the 'standard'. psutil is a well developed, cross-platform library which exactly addresses one need of the 'conditional PyCell compilation' feature.
I would not switch to a virtualenv, see also https://www.klayout.de/forum/discussion/2557/klayout-install-on-linux-system-python-vs-venv.
Nevertheless I've reduce the psutil dependency to the use of the feature, see https://github.com/IHP-GmbH/IHP-Open-PDK/pull/230

KrzysztofHerman, 2024-10-29:
@dnltz @ThomasZecha can we close this issue ?

ThomasZecha, 2024-10-29:
From my pov it can be closed.

dnltz, 2024-10-31:
I'm not complaining about `psutils` itself. It's indeed a well-maintained project. 

The problem is you should not install a Python package system-wide on your Linux machine. Actually Ubuntu24 kind-of blocks that now...

```
$ pip3 install psutils
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

I have multiple Python projects on my machine installed with each different tool requirements. Installing them all system-wide would end in a mess. You also want to make sure everyone is using the same version to lower support.

In my opinion: there should be a `requirements.txt` file in the project root directory and documentation how to create and use a virtualenv in the documentation.

ThomasZecha, 2024-11-01:
Pleae make a proposal for _'there should be a requirements.txt file in the project root directory and documentation how to create and use a virtualenv in the documentation'_

dnltz, 2024-11-01:
Sure, @ThomasZecha can you explain when `psutils` is actually required? I figured out I can use the PDK for digital designs without it, after you moved the function. Would help me to write documentation then.

ThomasZecha, 2024-11-04:
_psutils_ will be used if one of the PyCells loaded by https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/__init__.py uses the _#ifdef name_ instruction for conditional compilation, see also https://github.com/IHP-GmbH/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.tech/klayout/#readme

PR #227 (2024-10-18): ihp-sg13g2: libs.tech: klayout: python: Move getProcessNames()
This function has a hard depdendency to 'psutil' but is only required when IHP_PYCELL_LIB_PRINT_PROCESS_TREE is set. Move 'getProcessNames' inside the if block to allow using this PDK without 'psutil' installed.

Fixes #226

- [ ] Tests pass
- [ ] Appropriate changes to README are included in PR
PR #230 (2024-10-23): Fix https://github.com/IHP-GmbH/IHP-Open-PDK/issues/226
-reduced psutil-dependency to the use of conditional PyCell compilation

Fixes #226 >
closed 2024-11-12 by KrzysztofHerman