IHP-GmbH/IHP-Open-PDK #232: Xyce hv_pmos_id_vd_vg.cir;   says: error in sg13g2_moshv_mod.lib .. Unrecognized D for YPSP103_VA!M1

olisnr, 2024-10-21:
## Environment
```
System:
  Kernel: 6.8.0-47-generic x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 6.0.4 tk: GTK 3.24.33
    wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy
```
`Xyce DEVELOPMENT-202409220015-(Public_Release-7.8.0-128-g5b74b0d5)-opensource
`

## Expected Behavior
```
ich@x13g3:~/share/pdk/ihp-sg13g2/libs.tech/xyce/examples$ Xyce npn13G2_gain.cir
ich@x13g3:~/share/pdk/ihp-sg13g2/libs.tech/xyce/examples$ Xyce hv_pmos_id_vd_vg.cir
```
should simulate a .ac and a .dc example

## Actual Behavior
the .ac example runs, but the .dc example shows strange errors:

```
ich@x13g3:~/share/pdk/ihp-sg13g2/libs.tech/xyce/examples$ Xyce hv_pmos_id_vd_vg.cir

*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce DEVELOPMENT-202409220015-(Public_Release-7.8.0-128-g5b74b0d5)-opensource
***** Date: Mon Oct 21 23:26:32 CEST 2024

***** Executing netlist hv_pmos_id_vd_vg.cir

***** Reading and parsing netlist...
Netlist error in file ../models/sg13g2_moshv_mod.lib at or near line 87
 Unrecognized parameter D for device YPSP103_VA!M1
Netlist error in file ../models/sg13g2_moshv_mod.lib at or near line 87
 Unrecognized fields for device YPSP103_VA!M1
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 2 MSG_ERROR
 errors
*** Xyce Abort ***
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 2 MSG_ERROR
 errors
```

## Steps to Reproduce the Problem

run the simulations on my computer :)

i have no idea what could be wrong. my Xyce version is build from source and may be there is a Xyce ubuntu-package, that dont have the problem, but i didnt found it.

olisnr, 2024-10-22:
Correction: its not about .ac and .dc simulation. its about BJT or MOS-model... the MOS has the problem...

olisnr, 2024-10-22:
Update:
after reinstalling Xyce as described in 
`https://ihp-open-pdk-docs.readthedocs.io/en/main/analog/xyce.html 
`the error still remains the same.

dwarning, 2024-10-22:
The problem is the missing shared lib for the model given by -plugin after Xyce cmd. 

That's mine:

Xyce -plugin ~/Projects/spice/Verilog-A/VA-Models/admslibs/Xyce_Plugin_PSP103_VA.so hv_pmos_id_vd_vg.cir

olisnr, 2024-10-22:
great! thanks a lot.
did i miss the documentation or should this be written somewhere? like as in the `~/share/pdk/ihp-sg13g2/libs.tech/xyce/adms/README.md` or in the `examples` directory?
and also in
`https://ihp-open-pdk-docs.readthedocs.io/en/main/analog/xyce.html` ?

KrzysztofHerman, 2024-10-23:
@olisnr please checkout this branch and README
https://github.com/KrzysztofHerman/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.tech/xyce/adms
It is not yet part of the official repository but rather WiP.

olisnr, 2024-10-23:
@KrzysztofHerman thanks. i would also put it in the documentation:
https://ihp-open-pdk-docs.readthedocs.io/en/main/analog/xyce.html

closed 2024-10-23 by olisnr