IHP-GmbH/IHP-Open-PDK #234: two Xyce example questions

olisnr, 2024-10-23:
the code of the example hv_nmos_id_vg_vb.cir is:
```
psp103 test circuit
*
vd  d 0 dc 0.05
vg  g 0 dc 0.0
vs  s 0 dc 0.0
vb  b 0 dc 0.0
X1  d g s b sg13_hv_nmos w=2u l=0.5u mult=1

.option temp=27
.step  vb -3 0 0.25
.dc vg 0 3 0.02
.print dc format=gnuplot v(g) i(vd)

.lib ../models/cornerMOShv.lib mos_tt

.end
```

what does `psp103 test circuit`? and shouldnt the `mult=1` of `X1` be `m=1`?

KrzysztofHerman, 2024-10-23:
@olisnr the instance entry for the devices is the same for ngspice and xyce so
```
XM1 d g s b sg13_hv_nmos W=1u L=1u ng=2 m=1
```

olisnr, 2024-10-23:
then the examples should be changed. 

an other question is, there are no `cornerCAP.lib` and `cornerRES.lib` files in the `xyce/models` folder can i use this libraries from ngspice in xyce?

KrzysztofHerman, 2024-10-24:
@olisnr since `Xyce` support is under development please checkout my fork. 
https://github.com/KrzysztofHerman/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.tech/xyce/models
We are about to merge it to the main repo so any feedback  is more than welcome. 
Be careful with the paths to the libraries, you have to provide a full path. 
Since there are a slight differences between the libraries for `ngspice` and `Xyce` now we will maintain two separate version of each.

olisnr, 2024-10-24:
the `model=cap_cmim` seems to work :ok_hand:

KrzysztofHerman, 2024-10-28:
@olisnr can we close?  Are there any other questions ?

closed 2024-10-28 by olisnr