IHP-GmbH/IHP-Open-PDK #239: SRAM macros failing LVS

urish, 2024-10-28:
## Environment

IHP-Open-PDK commit 860268eaa497aff29195bf380dabdcfdc69a5cff

## Expected Behavior

SRAM macros pass LVS checks

## Actual Behavior

SRAM macros fail LVS checks. I tested with the following macros:

- RM_IHPSG13_1P_1024x8_c2_bm_bist.gds
- RM_IHPSG13_1P_1024x64_c2_bm_bist.gds
- RM_IHPSG13_1P_64x64_c2_bm_bist.gds
- RM_IHPSG13_1P_4096x8_c3_bm_bist.gds

The errors seem to be related to `RSC_IHPSG13_CDLYX1_DUMMY` - I'm attaching a copy of the LVS run dir with the extracted circuit and the .lvsdb file:

[lvs_run_2024_10_28_12_29_51.zip](https://github.com/user-attachments/files/17542376/lvs_run_2024_10_28_12_29_51.zip)

## Steps to Reproduce the Problem

Check out this repo (dev branch), set `PDK_ROOT` to point at the checked out repo, and then run:

```bash
cd $PDK_ROOT/ihp-sg13g2/libs.tech/klayout/tech/lvs

python run_lvs.py --layout $PDK_ROOT/ihp-sg13g2/libs.ref/sg13g2_sram/gds/RM_IHPSG13_1P_1024x8_c2_bm_bist.gds --netlist $PDK_ROOT/ihp-sg13g2/libs.ref/sg13g2_sram/cdl/RM_IHPSG13_1P_1024x8_c2_bm_bist.cdl
```

sergeiandreyev, 2024-10-29:
Hi @urish, for the SRAM cells the user should set additional parameters for the LVS run, with default settings it will not work
we have to figure out if it's possible with the current implementation of the KLayout LVS
@atorkmabrains, @FaragElsayed2, can we discuss this?

sergeiandreyev, 2024-10-30:
> Thanks! Is there a way to tell LVS to skip the SRAM cells?

I'll check with Mabrains team..
I double checked the SRAM LVS using our commercial tools and the status is:
the following have LVS mismatches (UNMATCHED SCHEMATIC PIN LABELS): 1024x8, 1024x16, 4096x8, 4096x16
all others are clean

we will review the issues if these are critical or could be waived

atorkmabrains, 2024-10-30:
`blank_circuit` command should be able to do that for you. It has to be included in the rule deck.

urish, 2024-10-31:
Thanks, unfortunately, blank_circuit seems to confuse KLayout - when enabling it for RM_IHPSG13_1P_1024x8_c2_bm_bist, suddenly, there are many mismatched nets / devices at the top level (before they were matched correctly).

```
schematic.blank_circuit('RM_IHPSG13_1P_1024X8_C2_BM_BIST')
netlist.blank_circuit('RM_IHPSG13_1P_1024x8_c2_bm_bist')
```

Here's variation, which only tries to blacklist the mismatching cells inside the SRAM macro:

```
blank_circuit("RSC_IHPSG13_CDLYX1_DUMMY")
blank_circuit("RSC_IHPSG13_CINVX2")
blank_circuit("RSC_IHPSG13_CINVX8")
blank_circuit("RSC_IHPSG13_FILLCAP4")
blank_circuit("RSC_IHPSG13_FILLCAP8")
```

It is almost working - for reason, we're still getting mismatches for the pins of RSC_IHPSG13_FILLCAP4 and RSC_IHPSG13_FILLCAP8:

[image]

I'm attaching the GDS, CDL and the LVS script for this variation. I'm running it as follows:

```
klayout -b -rd in_gds=6_final.gds -rd cdl_file=6_final_concat.cdl -rd report_file=6_lvs.lvsdb -rd run_mode=deep -r sg13g2_full.lylvs
```

[issue-239.zip](https://github.com/user-attachments/files/17584211/issue-239.zip)

atorkmabrains, 2024-11-01:
@sergeiandreyev Anything is required from us?

sergeiandreyev, 2024-11-01:
Hi @atorkmabrains, yes, I will send an email

KrzysztofHerman, 2024-12-18:
@atorkmabrains a gentle reminder on this action

atorkmabrains, 2024-12-18:
@KrzysztofHerman @sergeiandreyev We could discuss further in a meeting.

conorKotwasinski, 2026-10-27:
I have a focused SRAM bitcell dimension question related to #239. In both
`RM_IHPSG13_1P_256x32_c2_bm_bist` and
`RM_IHPSG13_1P_512x32_c2_bm_bist`, the raw GDS bitcell has three resistor
markers: two on Metal2/29 and one on Metal3/29. Intersecting each marker with
its actual conductor gives a 0.60 x 0.20 um rectangle. The CDL bitcell has
three `lvsres` declarations with `l=6e-07 w=2.6e-07` (0.60 x 0.26 um).

I see this at pinned revision `2bbec755dc67ca3db0261c3d6163e15735d66710` and
upstream `c6562037497437ce6bc689e633e5b45abf9bbccb`. The four GDS/CDL files
are byte-identical between these revisions. This also affects CMOS5L's
shared SRAM library.

A small Python/KLayout reproducer reads only those four raw files, verifies
their hashes, and reports the geometry and CDL statements. It uses no
modified LVS deck. Its comparison is of dimension multisets, not device
connectivity. KLayout 0.29.12 and 0.30.9 produce the same measurements.

Is 0.26 um intentional for the `lvsres` model despite the 0.20 um geometry?
If so, is there a supported comparison setup or documentation for that
interpretation? Otherwise, which supplied view should be corrected? The
published LVSRES table lists W and L as comparison parameters. I have not
changed either supplied view or treated a diagnostic match as sign-off.

Related documentation:
https://ihp-open-pdk-docs.readthedocs.io/en/latest/verification/lvs/04_05_res.html#lvsres

<details>
<summary>Standalone Python/KLayout reproducer</summary>

Install `klayout==0.30.9` in a disposable Python environment. Save this as `inspect_bitcell_width_v1.py`, then run `python inspect_bitcell_width_v1.py --download --output inspection`. The output directory must be new. Four raw public data files are hash-checked; no LVS deck or PDK file is modified.

```python
#!/usr/bin/env python3
# SPDX-License-Identifier: Apache-2.0
"""Measure raw SRAM resistor geometry and CDL dimensions; this is not LVS.

Requires Python 3 and klayout.db. Downloads are data only, from the pinned
IHP repository, with mandatory byte hashes. No PDK rule deck is changed or run.
"""
import argparse
from datetime import datetime, timezone
from decimal import Decimal
import hashlib
import json
from pathlib import Path
import re
import urllib.request

import klayout.db as kdb

PIN = '2bbec755dc67ca3db0261c3d6163e15735d66710'
CURRENT = 'c6562037497437ce6bc689e633e5b45abf9bbccb'
EXPECTED = {
    256: {
        'gds': 'e77a4d177e02541fe6c58351008455f9a05a715b504f8e318599f3d41bf5f959',
        'cdl': 'f8989b3a4cc392c57cfafdf82985e27b1e6626b0948b6a926ca6ace2fe01531f',
    },
    512: {
        'gds': 'c0da50c3e63190f49776e0f5127ae241500605f2f6c9cf6000b625a0772560d1',
        'cdl': '8c12371fc9839e797753ac9c228fca30ae03d3c0c7252b33dd4bc882f341f204',
    },
}

def require(condition, message):
    if not condition:
        raise ValueError(message)

def sha(path):
    return hashlib.sha256(Path(path).read_bytes()).hexdigest()

[... 117 lines cut ...]
                    data = response.read(4 * 1024 * 1024 + 1)
                require(len(data) <= 4 * 1024 * 1024, 'Unexpected source size')
                path.write_bytes(data)
            digest = sha(path)
            require(digest == EXPECTED[size][extension], f'Source hash mismatch: {name}')
            inputs.append(dict(name=name, sha256=digest, url=url, downloaded=args.download))
            files[extension] = path
        results.append(inspect(files['gds'], files['cdl'], size))
    checks = controls()
    for row in inputs:
        require(sha(sources / row['name']) == row['sha256'], 'Raw input changed while inspecting')
    record = dict(completed=True, scope='Raw bitcell geometry/CDL dimension comparison; not LVS or sign-off',
                  klayout_version=kdb.__version__, ref=args.ref, source_files=inputs,
                  cases=results, measurement_controls=checks, script_sha256=sha(__file__),
                  foundry_authority_resolved=False, pdk_modified=False,
                  recorded_utc=datetime.now(timezone.utc).isoformat())
    (args.output / 'report.json').write_text(json.dumps(record, indent=2) + '\n')
    for row in results:
        print(json.dumps(dict(size=row['size'], comparison=row['dimension_multiset_comparison'])), flush=True)
    print(args.output / 'report.json', flush=True)

if __name__ == '__main__':
    main()
```
</details>

sergeiandreyev, 2026-10-27:
Hi @conorKotwasinski, thank you for reporting!
most likely it's a bug not catched during LVS and should be fixed, see my comment in https://github.com/IHP-GmbH/IHP-Open-PDK/pull/1121#issuecomment-6448497832