IHP-GmbH/IHP-Open-PDK #240: is there a IHP130nm.cfg for pygmid?

olisnr, 2024-10-28:
in `https://github.com/IHP-GmbH/IHP-Open-PDK-docs/blob/main/docs/analog/pygmid.rst` there is a command `python3 -m pygmid --mode sweep --config ~./tests/pygmid/sweep/IHP130nm.cfg` for generation of LUT and plots for the "gm/ID methodology".

but i didnt find the `IHP130nm.cfg` in the openPDK. where can i find it?

KrzysztofHerman, 2024-10-28:
@olisnr 
the configuration is here: https://github.com/dreoilin/pygmid/blob/ngspice/tests/pygmid/sweep/IHP130nm.cfg
Please be careful with the lookup tables, it can contain errors. 
BTW for this kind of short questions you can use our Slack channel
https://open-source-silicon.slack.com/archives/C0549N94FD3

olisnr, 2024-10-28:
thanks a lot

closed 2024-10-28 by olisnr