IHP-GmbH/IHP-Open-PDK #243: Error in Qucs-s example simulation

joerdsonsilva, 2024-10-29:
## Environment
OS/Platform: linux
Qucs-s 24.3.2
NGSpice 43

## Expected Behavior
The example circuits in the IHP-Open-PDK-SG13G2-Examples_pjr folder are expected to be simulated.

## Actual Behavior
The example circuits (ac_mim_cap.sch, dc_diode_op.sch, dc_hbt_13g2.sch, ..... ) do not simulate and generate an error message:
"Ngspice started...Note: No ".plot", ".print", or ".fourier" lines; no simulations run
Note: Compatibility modes selected: hs a

Circuit: * qucs 24.3.2 /home/joerdson/qucsworkspace/ihp-open-pdk-sg13g2-examples_prj/ac_mim_cap.sch
Simulation finished"

## Steps to Reproduce the Problem

1. Open Qucs-s
2. Main Dock -> Projects -> IHP-Open-PDK-SG13G2-Examples_pjr
3. Content -> Schematics -> Choose example

The component symbols are all square, maybe that's a problem too.
[Images.zip](https://github.com/user-attachments/files/17559930/Images.zip)

KrzysztofHerman, 2024-10-30:
@joerdsonsilva thank you for reporting this. 
Have you executed the installation script   `$PDK_ROOT/$PDK/libs.tech/qucs/install.py` ?
Verify if you have the symbol libraries installed under `~/.qucs/user_lib` or `~/Qucs_Workspace/user_lib`

joerdsonsilva, 2024-10-30:
Yes, I did. The problem was occurring in the second part, in the "Add Path With Subfolders" step. But the SG13G2 PDK library folder was found at:

/home/Joerdson/QucsWorkspace/IHP-Open-PDK/ihp-sg13g2/libs.tech/qucs/user_lib

With that, the problem was solved.

closed 2024-10-30 by joerdsonsilva

sergeiandreyev, 2024-11-18:
Hi @joerdsonsilva, could you please check the proposed solution?