IHP-GmbH/IHP-Open-PDK #246: NW.b violation when two ebufn_2 abutt themselves

smunaut, 2024-10-30:
See attached image.

[image]

The rule is 0.62 min, but with two ebufn_2 cells next to each other you get 0.42. And actually if one of them is in the other orientation it can get down to 0.345.

smunaut, 2024-10-30:
And as a side note, that rule isn't checked by the DRC deck.

sergeiandreyev, 2024-10-30:
> And as a side note, that rule isn't checked by the DRC deck.

yes, the rule `NW.b` is still not implemented:
 https://ihp-open-pdk-docu.readthedocs.io/en/latest/verification/drc/03_missing_rules.html

smunaut, 2024-10-30:
I added this locally, which does report those errors, but given it's mostly just a cut&paste of similar rules in `Mn.b` and I barely know what I'm doing, I'll let the DRC deck experts actually add it :grin: 

```diff
--- sg13g2_maximal.lydrc	2024-10-30 10:51:23.044619042 +0000
+++ sg13g2_maximal_nwb.lydrc	2024-10-30 10:51:09.123354890 +0000
@@ -1613,6 +1613,9 @@
     NWell_NW_a.dup
 end.().output("NW.a", "Min. NWell width = 0.62")
 -&gt; do
+    NWell.ext_fast_space(0.62.um, consider_intersecting_edges: false, ignore_non_axis_aligned_edges: true)
+end.().output("NW.b", "Min. NWell space or notch = 0.62")
+-&gt; do
     NWell.ext_fast_separation(NActHV_ana, 0.62.um)
 end.().output("NW.d1", "Min. NWell space to external N+Activ inside ThickGateOx = 0.62")
 -&gt; (;x, y) do
```

akrinke, 2024-10-30:
Yes, as a temporary fix, you can implement the rule like this. However, the rule should only look at connected polygons – that's why we haven't implemented it, yet.

sergeiandreyev, 2024-10-30:
FYI, #247

sergeiandreyev, 2024-11-28:
Hi @smunaut, could you please confirm that the update fixes the issue on your side as well.

closed 2024-12-18 by KrzysztofHerman