IHP-GmbH/IHP-Open-PDK #247: Need automatic generation of layout assembly (simple in-line, all with all) to run abutment checks for stdcells

sergeiandreyev, 2024-10-30:


smunaut, 2024-10-30:
FWIW I had done something similar for sky130 to find that kind of issue. I quickly hacked it for sg13g2
[try_cells.zip](https://github.com/user-attachments/files/17575176/try_cells.zip)

It tries every cell against every cell in different orientations.
It doesn't create continuous rows, but at least for me that was enough to run through DRC

sergeiandreyev, 2024-10-30:
@smunaut, great! thank you!

smunaut, 2024-10-30:
[drc.xml.zip](https://github.com/user-attachments/files/17576420/drc.xml.zip)

Attached in the DRC maximal results from the klayout deck on the test gds generated by the scripts.

* There are the `NW.b` known from `ebufn_2`.
* There is a few `CntB.a1` those are false positives. The script placed the columns too close and they touched. I didn't re-run because it takes like 2h to run the klayout drc deck ...
* There are a bunch of `M1.b` and those look like real potential issues.

akrinke, 2024-11-01:
@smunaut Thank you! I've added the layout to our regression test suite.

sergeiandreyev, 2024-11-19:
Hi @smunaut, I've added the script and the test GDS to the OpenPDK repo
also you to the AUTHORS 😉

closed 2024-11-28 by sergeiandreyev