IHP-GmbH/IHP-Open-PDK #249: Can not run HBT example from Qucs-s using Xyce simulator

rupokdas147, 2024-10-31:
## Environment
Qucs-s 24.3.1
Xyce 7.8

## Expected Behavior
The examples given in "sg13g2/libs.tech/qucs-s/exampels" does not run using Xyce.
But the examples given in "sg13g2/libs.tech/xyce/examples" run when I run the cir files from Terminal, and I can plot them using Gnuplot.
Can anyone help me how can I run schematics from Qucs-s using Xyce?

## Actual Behavior
[image]

## Steps to Reproduce the Problem

1. Open Qucs-s.
2. Projects > IHP-Open-PDK-SG13G2-Examples_prj > Schematics > dc_hbt_13g2.sch
3. Change Simulator to Xyce.
4. Change Model Library Directory from Ngspice to Xyce.
5. Simulate

rupokdas147, 2024-11-01:
@KrzysztofHerman Here is the Netlist file...

* Qucs 24.3.0  /home/rf/QucsWorkspace/IHP-Open-PDK-SG13G2-Examples_prj/dc_hbt_13g2.sch

.SUBCKT IHP_PDK_nonlinear_components_npn13G2  gnd c b e bn t Nx=1  
.INCLUDE   ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/xyce/models/sg13g2_hbt_mod.lib
X1 c b e bn t  npn13G2 Nx={Nx} 
.ENDS
  
.LIB /home/rf/IHP-Open-PDK/ihp-sg13g2/libs.tech/xyce/models/cornerHBT.lib hbt_typ
VPr1 _net0 _net1 DC 0
V2 _net0 0 DC 1
I1 0 _net2 DC 0

Xnpn13G2 0  _net1 _net2 0 gnd IHP_PDK_nonlinear_components_npn13G2 Nx=1
.dc v2 0 1.5 0.005
.step i1 0 5e-06 5.55556e-07
.PRINT  dc format=raw file=spice4qucs.dc._swp.plot I(VPr1) 
.END

rupokdas147, 2024-11-01:
This line automatically becomes ngspice. Don't know why

* Qucs 24.3.0  /home/rf/QucsWorkspace/IHP-Open-PDK-SG13G2-Examples_prj/dc_hbt_13g2.sch

.SUBCKT IHP_PDK_nonlinear_components_npn13G2  gnd c b e bn t Nx=1  
.INCLUDE   ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech**/ngspice/**models/sg13g2_hbt_mod.lib
X1 c b e bn t  npn13G2 Nx={Nx} 
.ENDS
  
.LIB /home/rf/IHP-Open-PDK/ihp-sg13g2/libs.tech/xyce/models/cornerHBT.lib hbt_typ
VPr1 _net0 _net1 DC 0
V2 _net0 0 DC 1
I1 0 _net2 DC 0

Xnpn13G2 0  _net1 _net2 0 gnd IHP_PDK_nonlinear_components_npn13G2 Nx=1
.dc v2 0 1.5 0.005
.step i1 0 5e-06 5.55556e-07
.PRINT  dc format=raw file=spice4qucs.dc._swp.plot I(VPr1) 
.END

rupokdas147, 2024-11-01:
I changed every "ngspice" to xyce and tried to run it from terminal using "xyce [netlist]" command. Here are the errors:

*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce Release 7.6-opensource
***** Date: Fri Nov 01 06:28:17 EDT 2024

***** Executing netlist /home/rf/QucsWorkspace/IHP-Open-PDK-SG13G2-Examples_prj/dc_hbt_13g2.cir

***** Reading and parsing netlist...
Netlist error: Number of nodes for subcircuit instance XNPN13G2 does not agree
 with number of nodes in subcircuit IHP_PDK_NONLINEAR_COMPONENTS_NPN13G2
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 1 MSG_ERROR
 errors
*** Xyce Abort ***
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 1 MSG_ERROR
 errors

*** Xyce Abort ***
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.

KrzysztofHerman, 2024-11-01:
@rupokdas147  so there is an issue with the binding to a valid model. 
In this #241 PR we provide some solution for xschem and ngspice, which binds symbols to valid models.
This is still not working with Qucs-S 
As you can see this HBT is a 5 terminal device and the symbol is 4 terminal in Qucs-S so the `t` terminal is floating.
```
.SUBCKT IHP_PDK_nonlinear_components_npn13G2 gnd c b e bn t Nx=1
.INCLUDE ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib
X1 c b e bn t npn13G2 Nx={Nx}
.ENDS
```

You could try to copy the `sg13g2_hbt_mod.lib` file from the #241 PR and place it in  
` ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib`.  Then
use this netlist subckt to simulate 

```
.SUBCKT IHP_PDK_nonlinear_components_npn13G2 gnd c b e bn  Nx=1
.INCLUDE ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib
X1 c b e bn  npn13G2 Nx={Nx}
.ENDS
```
Validate also if you have the recent version of the PDK. 
Your symbol entry in `.qucs/user_lib/IHP_PDK_nonlinear_components.lib` should look like this:
```
<Component npn13G2>
  <Description>
Author: Mike Brinson, Feb 2024
mbrin72043@yahoo.co.uk, or m.brinson@londonmet.ac.uk.
  </Description>
  <Model>
.Def:IHP_PDK_nonlinear_components_npn13G2 c b e bn Nx="1" 
.Def:End
  </Model>
  <Spice>*
.SUBCKT IHP_PDK_nonlinear_components_npn13G2  gnd c b e bn Nx=1  
X1 c b e bn npn13G2 Nx={Nx} 
.ENDS
  </Spice>
  <VerilogModel>  </VerilogModel>
  <VHDLModel>  </VHDLModel>
  <Symbol>
  <Line 0 20 0 30 #ff0000 4 1>
  <Line 0 -50 0 30 #ff0000 4 1>
  <Line 0 -20 -10 10 #ff0000 4 1>
  <Line -10 -20 0 40 #ff0000 4 1>
  <Line -10 10 10 10 #ff0000 4 1>
  <Line -10 0 -20 0 #ff0000 4 1>
  <Line 10 0 -10 0 #ff0000 4 1>
  <.PortSym 10 0 4 180>
  <Arrow -10 10 10 10 6 3 #ff0000 3 1 1>
  <.PortSym 0 50 3 0>
  <.PortSym -30 0 2 0>
  <.PortSym 0 -50 1 0>
  <.ID 10 64 npn13G2 "1=Nx=1">
  </Symbol>
</Component>

```

rupokdas147, 2024-11-01:
@KrzysztofHerman The Qucs-s working fine with Ngspice. I want to design Power Amplifier. So I need large signal analysis, which doesn't support by Ngspice. I will try the solution you given above and let you inform if the Xyce is working or not in the next comment.

KrzysztofHerman, 2024-11-01:
The issue we are trying to solve with PR #241 is exact binding between a device  symbol and model. The `*.cir` netlists  with examples for Xyce use 5 terminal HBT device. The PR introduces two version of each device `npn13G2`  and `npn13G2_5t` where the difference is the number of terminals (additional `t` output which is correlated with device temperature and self heating ).

KrzysztofHerman, 2024-11-01:
I have just added xyce support for 4 and 5 terminal devices, See last commit in PR #241.
There is a difference in VBIC model level ngspice supports level 9 and Xyce level 12. We still do not have a 5 terminal symbol for HBT's in Qucs-S

rupokdas147, 2024-11-02:
@KrzysztofHerman If I change "IHP_PDK_nonlinear_components_npn13G2" to "npn13G2" the netlist simulates from terminal using Xyce.

`* Qucs 24.3.0  /home/rf/QucsWorkspace/IHP-Open-PDK-SG13G2-Examples_prj/dc_hbt_13g2.sch

.SUBCKT IHP_PDK_nonlinear_components_npn13G2  gnd c b e bn t Nx=1  
.INCLUDE   ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib
X1 c b e bn t  npn13G2 Nx={Nx} 
.ENDS
  
.LIB /home/rf/IHP-Open-PDK/ihp-sg13g2/libs.tech/xyce/models/cornerHBT.lib hbt_typ
VPr1 _net0 _net1 DC 0
V2 _net0 0 DC 1
I1 0 _net2 DC 0

Xnpn13G2 0  _net1 _net2 0 gnd IHP_PDK_nonlinear_components_npn13G2 Nx=1
.dc v2 0 1.5 0.005
.step i1 0 5e-06 5.55556e-07
.PRINT  dc format=raw file=spice4qucs.dc._swp.plot I(VPr1) 
.END
`

rupokdas147, 2024-11-02:
I don't know why Qucs-s adding the library name with the component name. I am actually a newbie in this sector. Removing the "IHP_PDK_nonlinear_components" from the above-highlighted line also helps to simulate the netlist in Ngspice. I have removed the exact phrase from a Ngspice generated Netlist and the netlist simulated in Ngspice smoothly.

KrzysztofHerman, 2024-11-04:
@rupokdas147 unfortunately this is the way you can define custom symbols in Qucs-S. You have to wrap it  using
subcircuit and additionally reference the spice library with the model of the device you use. In this particular case the device is `npn13G2`, the wrapper sub-circuit is `IHP_PDK_nonlinear_components_npn13G2` and finally the instance name is 
`Xnpn13G2`. There is a discussion on [Qucs-S discussion space](https://github.com/ra3xdh/qucs_s/discussions/876) where we discuss enhancements of Qucs-S so maybe you could drop there your suggestions if you have any.

KrzysztofHerman, 2024-12-18:
@rupokdas147 please update the PDK  and run your testcase again.

rupokdas147, 2024-12-20:
@KrzysztofHerman Thanks Worked

closed 2024-12-20 by rupokdas147