IHP-GmbH/IHP-Open-PDK #251: Transmission Line is missing.

rupokdas147, 2024-11-01:
You mentioned that the PDK is specially made for RF designs. But there are no Transmission Line for both simulation and layout. How can I use transmission line when designing the circuit? Moreover there are Inductors in Layout only, no Inductor found for simulation.

KrzysztofHerman, 2024-11-04:
@rupokdas147 as you can see the current status of the PDK is `Preview`.  The things you are mentioning are on our TODO list so we will provide it next year. Nevertheless we are trying to support any external contributors willing to provide any solution aligned with our development plans so if you would like to develop some RF components, model it tapeout and measure feel free to do so.  We have `OpenMPW` shuttles where you could try out your ideas.

rupokdas147, 2024-11-06:
@KrzysztofHerman I will try to create spice model and layout for the devices. Could you provide me some necessary information about the fabrication process like Layer Stack information, Process Design Rules, EM simulation guidelines etc?

KrzysztofHerman, 2024-11-06:
@rupokdas147 the Process Specification and Design Rule Manual are already on github. You can find it here [process spec](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.doc/doc/SG13G2_os_process_spec.pdf) and [drm](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.doc/doc/SG13G2_os_layout_rules.pdf)
As for EM simulation the basic information can be found [here](https://github.com/IHP-GmbH/IHP-Open-PDK/tree/main/ihp-sg13g2/libs.tech/openems). I also recommend you to subscribe to the [ihp-sg13g2 Slack channel ](https://join.slack.com/share/enQtODAxMDc3NjE0MDI4OC1jY2JmNDg2NjViZGYyNjNkMGUwZjk3OTAyNjU2YjhkMTUyZmYyOTk0YmRhNDI4YTliZGM2YjdmODE2YWY4ZGM1)

rupokdas147, 2024-12-20:
@KrzysztofHerman Thanks Worked