IHP-GmbH/IHP-Open-PDK #252: Use of the dummy liberty for I/O cells?

Syndace, 2024-11-03:
Does the file [sg13g2_io_dummy.lib](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.ref/sg13g2_io/lib/sg13g2_io_dummy.lib) still have a use considering there are full liberty models for the I/O cells now?

smunaut, 2024-11-03:
The others don't have the Analog pad entry for instance so when using the `.lib` with a synthesis tool it won't recognize the cell because it's not declared anywhere.

KrzysztofHerman, 2024-11-04:
@Syndace as @smunaut the Analog pad entry is only a placeholder to make the ORFS finish its flow. If you have a fully digital chip it is recommended to use the liberty views provided for each corner.

Syndace, 2024-11-04:
Thanks for the clarification!

closed 2024-11-04 by Syndace