IHP-GmbH/IHP-Open-PDK #254: LVS unexpectedly failing

stafverhaegen-chipflow, 2024-11-07:
## Environment

## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

1.
2.
3.

stafverhaegen-chipflow, 2024-11-07:
Sorry, pressed wrong button.

stafverhaegen-chipflow, 2024-11-07:
Updated the problem description with test case.

KrzysztofHerman, 2024-11-07:
@stafverhaegen-chipflow @atorkmabrains @FaragElsayed2 
I have just reproduced the testcase and indeed it fails in the command line but it seems to pass in the GUI
using the default settings:
[image]

The output contains some warnings:
[image]

The log view:
[image]

I have used `klayout 0.29.0` and `python 3.10.12`

sergeiandreyev, 2024-11-07:
shouldn't the batch-mode deck and GUI-mode deck behave the same? do they share the same implementation?

sergeiandreyev, 2024-11-07:
@KrzysztofHerman, could you please share here the error message from the batch mode LVS run?

KrzysztofHerman, 2024-11-07:
Here is the output from the command line run:
```
2024-11-07 13:37:44 +0100: Memory Usage (755784K) : ERROR : Netlists don't match
2024-11-07 13:37:44 +0100: Memory Usage (755784K) : xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
2024-11-07 13:37:44 +0100: Memory Usage (755784K) : LVS Total Run time 2.634887 seconds
07-Nov-2024 13:37:44 | INFO    | Total execution time 4.176420450210571s

***** C4M deck *****
LVS OK

```
GUI based run
```
2024-11-07 13:38:54 +0100: Memory Usage (1472000K) : ==========================================
2024-11-07 13:38:54 +0100: Memory Usage (1472000K) : INFO : Congratulations! Netlists match.
2024-11-07 13:38:54 +0100: Memory Usage (1472000K) : ==========================================
2024-11-07 13:38:54 +0100: Memory Usage (1472000K) : LVS Total Run time 3.395420 seconds

```

stafverhaegen-chipflow, 2024-11-07:
I tried again with main branch and now for me it does seem to match when running from command line in flat mode; e.g. add `--run_mode=flat`. Does it work from GUI in hierarchical mode ?

stafverhaegen-chipflow, 2024-11-07:
When I run the C4M deck also in deep mode it also fails. It thus seems that klayout has problems with this layout in deep mode. I don't see immediately something wrong with it.
I changed C4M deck to deep mode by chaging line `flat` to `deep` in file `c4m_klayout/share/C4M.IHPSG13G2.lvs` of test case.

sergeiandreyev, 2024-11-13:
Hi @stafverhaegen-chipflow, you have connectivity issues..
so if we say the layout is the reference, you should change the netlist from:
```
.subckt SP6TCArray_2X1 vss vdd1 vdd2 wl[0] wl[1] bl[0] bl_n[0]
Xinst0x0 wl[0] bl[0] bl_n[0] vdd1 vss SP6TCCell
Xinst1x0 wl[1] bl[0] bl_n[0] vdd2 vss SP6TCCell
.ends SP6TCArray_2X1
```
to
```
.subckt SP6TCArray_2X1 vss vdd1 vdd2 wl[0] wl[1] bl[0] bl_n[0]
Xinst0x0 wl[0] bl[0] bl_n[0] vdd2 vss SP6TCCell
Xinst1x0 wl[1] bl[0] bl_n[0] vdd1 vss SP6TCCell
.ends SP6TCArray_2X1
```

KLayout LVS should be clean (I would be very surprised if it will not be the case)
p.s. I checked on commercial tools

sergeiandreyev, 2024-11-13:
IMHO, from the KLayout LVS reports posted here it's nearly impossible to understand the issue root cause..

stafverhaegen-chipflow, 2024-11-14:
I was already checking with the change you proposed locally. It does not change anything. I made a new test case: [20241114_lvs_case.tar.gz](https://github.com/user-attachments/files/17747483/20241114_lvs_case.tar.gz)
Here the spice netlist was updated as you proposed, I updated the run.sh script so it runs both the IHP-Open-PDK LVS in flat and deep mode and also my generated deck. Both are OK in flat mode, both fail in deep mode. The output of the run is in the included run.log file. I'm not sure how I can make the test case more clear.

sergeiandreyev, 2024-11-14:
ok, I asked KLayout LVS deck developers to take a look on this

atorkmabrains, 2024-11-14:
@stafverhaegen-chipflow Just to confirm my understanding of the issue that you have, there is a test case that pass LVS if you run from GUI and fails if you run from command line.

- Case 1: Run GUI ----> Pass
- Case 2: Run cmd ---> Fail
- Case 3: Run cmd Flat ----> Pass
- Case 4: Run cmd Deep ---> Fail
- Case 5: Run GUI Deep ---> Fail

Could you please elaborate on what I  mentioned above? If there are more cases, could you please list them the same way I mentioned above. That would really help us review quickly the root cause.

atorkmabrains, 2024-11-14:
@stafverhaegen-chipflow

Fundamentally, it will fail if deep run and pass if it's flat regardless of the way of invocation. Is that correct?

atorkmabrains, 2024-11-14:
@stafverhaegen-chipflow 

One last question, Did you run this test case on other tool (Commercial) and it passed?

sergeiandreyev, 2024-11-14:
> Hi @stafverhaegen-chipflow, you have connectivity issues.. so if we say the layout is the reference, you should change the netlist from:
> 
> ```
> .subckt SP6TCArray_2X1 vss vdd1 vdd2 wl[0] wl[1] bl[0] bl_n[0]
> Xinst0x0 wl[0] bl[0] bl_n[0] vdd1 vss SP6TCCell
> Xinst1x0 wl[1] bl[0] bl_n[0] vdd2 vss SP6TCCell
> .ends SP6TCArray_2X1
> ```
> 
> to
> 
> ```
> .subckt SP6TCArray_2X1 vss vdd1 vdd2 wl[0] wl[1] bl[0] bl_n[0]
> Xinst0x0 wl[0] bl[0] bl_n[0] vdd2 vss SP6TCCell
> Xinst1x0 wl[1] bl[0] bl_n[0] vdd1 vss SP6TCCell
> .ends SP6TCArray_2X1
> ```
> 
> KLayout LVS should be clean after this update (I would be very surprised if it will not be the case) p.s. I checked on commercial tools

@atorkmabrains, I ran it with commercial tools - there is indeed an LVS issue and I showed how to make it clean

stafverhaegen-chipflow, 2024-11-14:
> @stafverhaegen-chipflow
> 
> Fundamentally, it will fail if deep run and pass if it's flat regardless of the way of invocation. Is that correct?

Indeed

atorkmabrains, 2024-11-14:
> @atorkmabrains, I ran it with commercial tools - there is indeed an LVS issue and I showed how to make it clean

@sergeiandreyev I presume you ran it hierarchical in the commercial tool, correct? Did you use "Automatch" option in the commercial tool as well to see which exact cell needs changing?

I understand that you have provided a change required for the netlist, @stafverhaegen-chipflow Did you test that change and ran LVS in deep mode?

atorkmabrains, 2024-11-14:
@sergeiandreyev That's automatch switch.

sergeiandreyev, 2024-11-19:
Hi @stafverhaegen-chipflow, as I understand the issue is related to this one opened on KLayout:
https://github.com/KLayout/klayout/issues/1921

stafverhaegen-chipflow, 2024-11-26:
I doubt this is related to that issue. In that issue LVS has a problem due to ambiguity between different floating nets but our test case should not have ambiguity. Also in that issue the problem is with flat LVS and in out case the problem is with deep LVS.

stafverhaegen-chipflow, 2024-11-26:
It may be related to KLayout/klayout#1813 though. My thesis is that `extend("*")` is not doing what it is expecting to do for the given test case. Will look a little further now.

stafverhaegen-chipflow, 2024-11-27:
Created KLayout/klayout#1935 now with slightly adapted test case using global net and not `extend("*")`. I assume solution for that will also be applicable for the IHP LVS deck.

FaragElsayed2, 2024-12-18:
@sergeiandreyev @KrzysztofHerman 

I’ve investigated the `RM_IHPSG13_1P_64x64_c2_bm_bist.gds` file using deep run mode, and here are my findings:

1. For the `RSC_IHPSG13_CDLYX1_DUMMY` cell, it uses the `lvsres` resistor, but its name does not match the one implemented in LVS.  
   [image]

2. The `RSC_IHPSG13_WLDRVX8` cell has a similar issue where devices in its cells are not being extracted (likely due to a hierarchy problem). I believe this will be resolved in a flat mode run. Refer to #254.

3. Some cells exist in the CDL but are missing in the layout. I suspect these might exist at the top level in the layout, so this issue may also be resolved in a flat run.  
   [image]

CC @atorkmabrains