IHP-GmbH/IHP-Open-PDK #264: KLayout keybindings

hpretl, 2024-11-16:
In order to be efficient in layout we need more keybindings for the most used operations in KLayout. I am not sure what would be a good way to bring them in. I guess we could add Python code to the already existing autorun code to set them, or better yet, as people might want to adapt them to their liking, provide a mechanism to easily set keybindings. Maybe this should be a direct easy functionality in KLayout itself to specify keybindings.

hpretl, 2024-11-16:
See https://github.com/KLayout/klayout/issues/1926

hpretl, 2024-11-18:
It looks like `klayoutrc` is the proper way to implement keybindings, which are then user specific and PDK agnostic.

KrzysztofHerman, 2024-11-18:
@hpretl correct, the place where the user configuration goes is this `klayoutrc` file. Reviewing it one can found account specific fixed path names so distributing  a generic one does not make a lot of sense.  Maybe providing only the section `<key-bindings> </key-bindings>`   would be useful.

hpretl, 2024-11-25:
BTW, this is what we now use in our IIC-OSIC-TOOLS (from version `2024.11` on):
```
<?xml version="1.0" encoding="utf-8"?>
<config>
 <default-grids>1.0,0.1,0.01,0.005</default-grids>
 <edit-snap-objects-to-grid>true</edit-snap-objects-to-grid>
 <edit-snap-to-objects>true</edit-snap-to-objects>
 <grid-micron>0.005</grid-micron>
 <key-bindings>
'@secrets.duplicate_interactive':C;
'@secrets.paste_interactive':H;
'@secrets.sel_move_interactive':M;
edit_menu.cell_menu.adjust_cell_origin:O;
edit_menu.clear_all_rulers:'Shift+K';
edit_menu.edit_options:F3;
edit_menu.mode_menu.box:B;
edit_menu.mode_menu.instance:I;
edit_menu.mode_menu.partial:S;
edit_menu.mode_menu.path:P;
edit_menu.mode_menu.polygon:'Shift+P';
edit_menu.mode_menu.ruler:R;
edit_menu.mode_menu.text:X;
edit_menu.redo:'Ctrl+Z';
edit_menu.select_menu.select_all:'Ctrl+Shift+A';
edit_menu.select_menu.unselect_all:'Ctrl+D';
edit_menu.selection_menu.align:A;
edit_menu.selection_menu.change_layer:'Shift+L';
edit_menu.selection_menu.flatten_insts:'Alt+Shift+F';
edit_menu.selection_menu.make_cell:'Alt+Shift+M';
edit_menu.selection_menu.sel_flip_x:'Shift+H';
edit_menu.selection_menu.sel_flip_y:'Shift+V';
edit_menu.selection_menu.sel_move:'[';
edit_menu.selection_menu.sel_move_to:']';
edit_menu.selection_menu.sel_rot_ccw:'Shift+R';
edit_menu.selection_menu.sel_rot_cw:'Ctrl+Shift+R';
edit_menu.selection_menu.tap:T;
edit_menu.selection_menu.union:'Shift+M';
edit_menu.undo:U;
file_menu.new_layout:'Ctrl+N';
file_menu.open_same_panel:'Ctrl+Shift+O';
file_menu.save:'Ctrl+S';
file_menu.save_as:'Ctrl+Shift+S';
file_menu.reload:'Ctrl+Shift+R';
help_menu.assistant:F1;
view_menu.crosshair_cursor:'Shift+C';
zoom_menu.ascend:'Ctrl+E';
zoom_menu.descend:'Shift+E';
zoom_menu.descend_into:E;
zoom_menu.max_hier:'Shift+F';
zoom_menu.redraw:'Ctrl+R';
zoom_menu.select_cell:'Shift+S';
zoom_menu.select_current_cell:'Shift+T';
zoom_menu.zoom_fit:F;
zoom_menu.zoom_fit_sel:'Ctrl+F';
zoom_menu.zoom_in:Return;
zoom_menu.zoom_out:'Shift+Return';
zoom_menu.max_hier_0:0;
zoom_menu.max_hier_1:9;
'@lcp_context_menu.hide_empty_layers':4;
'@lcp_context_menu.invvalid':6;
'@lcp_context_menu.show_all':1;
'@lcp_context_menu.show_only':2;
'@lcp_context_menu.toggle_visibility':3;
'@lcp_context_menu.valid':5
 </key-bindings>
</config>
```

sergeiandreyev, 2024-12-09:
Hi @hpretl, we're going to provide this example somewhere in the documentation, until then I propose to leave this issue open

KrzysztofHerman, 2024-12-10:
@hpretl @sergeiandreyev the Klayout configuration, including key-bindings is here 
https://ihp-open-pdk-docs.readthedocs.io/en/latest/analog/klayout.html
I will close this issue.

closed 2024-12-18 by KrzysztofHerman