IHP-GmbH/IHP-Open-PDK #265: Layout efficiency: Substrate tap, nwell tap, vias, improved MOSFET cells

hpretl, 2024-11-16:
Maybe I just don't know how to do it properly, but it looks like a few fundamental thing are still missing from the KLayout setup:

* Via generator (with selectable start and end layers, and number of X/Y multiples)
* Substrate contact (maybe merge into via generator); ideal would be the possibility to make a substrate contact as a path
* Option to NMOS pcell to add a guardings
* Nwell contact (maybe merge into via generator); ideal would be the possibility to make an nwell contact as a path
* Option to PMOS pcell to add a guardings

In addition, it would be good have the option in the MOSFET pcells to have the gates connected.

KrzysztofHerman, 2024-11-18:
@hpretl thank you for your comment. 

there is ongoing discussion of the via should be generated natively by the tool, Klayout in this case, or as a separate PyCell. Do you have any opinion on that ?

Substrate/Nwell contacts are among the devices that are planned to be delivered soon.

I have develop a  guard ring PyCell and I can share the code.  The feature to have it as a mosfet PyCell option we have to discuss internally.

hpretl, 2024-11-18:
> there is ongoing discussion of the via should be generated natively by the tool, Klayout in this case, or as a separate PyCell. Do you have any opinion on that ?

As a via/viastack is a fundamental component of any PDK, a KLayout implementation would definitely make sense. But given the current state where custom layout in KLayout is really cumbersome due to lack of vias any solution in a reasonable timeline would help. 😀

hpretl, 2024-11-18:
@KrzysztofHerman One more argument for having vias inside KLayout: Eventually I would like to have a path function where I can change layers while routing. For this functionality, we need vias, and probably everything is easier if vias are a native functionality of KLayout.

sergeiandreyev, 2024-11-18:
ok, will do

sergeiandreyev, 2024-11-19:
FYI, https://github.com/KLayout/klayout/issues/1931

sergeiandreyev, 2024-11-19:
> But given the current state where custom layout in KLayout is really cumbersome due to lack of vias any solution in a reasonable timeline would help.

@hpretl, maybe for the time being you can use the `via` Pycell (`via_stack`) from the old version of IHP OpenPDK as a temporary workaround:
```
> git checkout 8dbe642b8c0512d3c6b52feee1e3beac837b191e
```