IHP-GmbH/IHP-Open-PDK #267: Primitive device: PNP/NPN (vertical oder lateral) for CMOS-only

hpretl, 2024-11-16:
For CMOS-only designs, an NPN or PNP (either lateral or vertical) would be needed for bandgaps and similar uses.

hpretl, 2024-11-17:
Thanks for the pointer, you are absolutely right! I totally missed it because there is no pcell in KLayout (at least none that I can find) and no mention in the process description, but there is a model and symbol for simulation and a fixed layout in `sg13g2_pr.gds`.

@KrzysztofHerman @sergeiandreyev Are there plans to provide a pcell, or is there one fixed size of PNP with a fixed layout (which is quite the norm for parasitic PNPs)? In any case, a description in the `SG13G2_process_spec.pdf` would be great!

hpretl, 2024-11-18:
@sergeiandreyev cool! Do you have a rough timeline for the remaining cells?

sergeiandreyev, 2024-11-18:
unfortunately, we don't have dedicated resources for this task, so the timeline is rather hard to define..
I agree, it will be useful for tracking, so I will create a separate GitHub 'enhancement' issue for new devices

sergeiandreyev, 2024-11-18:
> In any case, a description in the `SG13G2_process_spec.pdf` would be great!

after checking this internally, the feedback is that we have a **new** `pnpMPA` device implementation in progress, so the process spec is planned to be updated with this **new** device

hpretl, 2024-11-18:
> after checking this internally, the feedback is that we have a new pnpMPA device implementation in progress, so the process spec is planned to be updated with this new device

Excellent, good to hear! @sergeiandreyev Do you have a timeline for the bundle (doc, model, pcell)?

sergeiandreyev, 2024-11-18:
> Do you have a timeline for the bundle (doc, model, pcell)?

around 6 month..

olisnr, 2024-11-24:
because You allready give us the measurements of different pnpMPAs, it would be nice, You also give the test-chip-layouts as` .gds` with it. then the need for a PCell is much smaller. 
at the moment there is only the 1u/2u variant aviablle from the `ihp-sg13g2/libs.ref/sg13g2_pr/gds/sg13g2_pr.gds`

olisnr, 2024-11-24:
im also interested in the SVaricap. is there more information about what it can and how to use?

RTimothyEdwards, 2024-12-07:
@sergeiandreyev :  I have a follow-up question about the pnpMPA device.  Looking at the GDS of the device in the `sg13g2_pr` GDS (because there is no klayout pycell for it), there is nothing that uniquely identifies this device as a bipolar, which makes it impossible to uniquely identify the structure as a bipolar transistor for the purpose of netlist extraction.  The simple expedient of allowing the bipolar ID layer (TRANS.drawing) to cover the device would allow it to be extracted.  The PNP device would be differentiated from the SiGe devices by recognizing that the TRANS layer does not contain any emitter window.  Of course, since this would be a non-mask layer, you could just define any arbitrary unused layer type for a PNP device recognition layer.

sergeiandreyev, 2025-11-13:
Hi @FaragElsayed2, could you please reply Tim how the `pnpMPA` recognition is done in KLayout LVS? I think we have this info in the documentation, but maybe it's better to provide more details..

RTimothyEdwards, 2025-11-13:
@FaragElsayed2 :  That is basically what I am doing to identify a PNP device.  Thanks for the confirmation.

But stil, it seems rather arbitrary, easy to think of geometry that would be identified as a PNP but which is not a PNP, and it would be trivially easy to add an identifier layer for PNPs (including just using the existing TRANS marker, as I indicated above) and would make the extraction process both less complicated, and faster, and eliminate any chance that layout might be interpreted incorrectly.  Just my two cents (and noting that yesterday, the U.S. stopped minting pennies, so my two cents is now valueless).

sergeiandreyev, 2025-11-13:
ok, I'll discuss this internally