IHP-GmbH/IHP-Open-PDK #271: LVS error after script Fill

joerdsonsilva, 2024-11-20:
## Environment
OS/Platform: linux
Klayout 0.29.7
NGSpice 43

## Expected Behavior
LVS clean after applying Fill to the design.

## Actual Behavior
When applying Fill to the design, the LVS presents an error. But the design without applying Fill does not present an error in the LVS.

## Steps to Reproduce the Problem

1. Open gds file with Fill in klayout
2. Run klayout LVS with cdl file
3. After simulation, the error will be displayed
4. To compare the results, you can repeat the steps with the NoFill file and observe the clean LVS.

[fill.zip](https://github.com/user-attachments/files/17826376/fill.zip)
[nofill.zip](https://github.com/user-attachments/files/17826377/nofill.zip)

olisnr, 2024-11-24:
@joerdsonsilva looks nice, what is the reason to change the layer all 100µm?

joerdsonsilva, 2024-11-24:
Please provide more details about your question. Did you check this error in the LVS of the file with fill?

[image]

olisnr, 2024-11-25:
it was only a question, because i dont see the reason to switch from layer M2 to M3 and back. may be its about thermal stress or so...

i dont see the reason for LVS problem. it seems that Your filling pattern are recognized as NFETs and all the bodies are connected to BN. but why i dont know. 

the only things i saw is, the PADs shouldnt have a TopVia2...

joerdsonsilva, 2024-11-25:
Got it, different layers of metal were used to avoid possible antenna effects. Just a precaution. 

Without Fill, this error does not generate in LVS, so it must be something in how the fill is recognized.

olisnr, 2024-11-25:
@joerdsonsilva its funny, if You delete the seal-ring and all the PADs, You get much fewer errors. may be You have also to connect all the BN with metal. i try this also

olisnr, 2024-11-25:
but now there are BNs missing in the normal NFETs:

[image]

KrzysztofHerman, 2024-11-27:
@atorkmabrains @FaragElsayed2 could you please take a look and evaluate the possibility to exclude `GatePoly.filler` over `Active.filler` from a recognition as a mos device. I have tested it and it is the root cause why LVS reports mismatch.

FaragElsayed2, 2024-11-27:
@KrzysztofHerman 
As suggested by @atorkmabrains, dummy transistors that are not connected to Metal1 have been excluded from LVS.

KrzysztofHerman, 2024-11-27:
Thank you @FaragElsayed2. After your changes I have clean LVS. @joerdsonsilva can you check it on your setup ? If it works for you also we can merge and close.

atorkmabrains, 2024-11-27:
@KrzysztofHerman BTW, this use case is very unique and @FaragElsayed2 has taken the best clean approach to make sure that we are not taking dummy transistors into account.

joerdsonsilva, 2024-11-27:
> @joerdsonsilvaem relação ao efeito da antena. não seria melhor ir do gate direto para M2 e depois voltar para M1?

Yes, it is a valid approach.

joerdsonsilva, 2024-11-27:
> Thank you @FaragElsayed2. After your changes I have clean LVS. @joerdsonsilva can you check it on your setup ? If it works for you also we can merge and close.

Yes, I have a clean LVS after the changes.

[image]

PR #277 (2024-11-27): Ignoring dummy trans from being recognized in LVS
## Ignoring dummy trans from being recognized in LVS

Fix: #271
closed 2024-11-27 by KrzysztofHerman