IHP-GmbH/IHP-Open-PDK #275: [DRC] Violating M2.b is sometimes not detected

hpretl, 2024-11-25:
We are currently working towards a tapeout, and had a situation where the DRC fail below has not been detected by the current KLayout DRC rules. `M2.b` is clearly violated.

[image]

hpretl, 2024-11-25:
@akrinke Can you please take a look?

sergeiandreyev, 2024-11-25:
Hi @hpretl, I have it flagged by the `minimal` rule deck:
[image]
Could you please attach your testcase?

hpretl, 2024-11-25:
@sergeiandreyev Here is the M2 layer of the chip. Maybe we used an outdated DRC runset? Let's see if the error is flagged on your end.

[tc_m2_fail.oas.gz](https://github.com/user-attachments/files/17909010/tc_m2_fail.oas.gz)

sergeiandreyev, 2024-11-25:
yep, it is found:
[image]
I'm using latest deck from `dev` branch:
https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.tech/klayout/tech/drc/sg13g2_minimal.lydrc

hpretl, 2024-11-25:
@sergeiandreyev Thanks for checking! We need to find out why we missed the fail. I propose to close this issue.

closed 2024-11-25 by hpretl