IHP-GmbH/IHP-Open-PDK #280: [qucs-s] additional parameters for primitive devices and callbacks support

sergeiandreyev, 2024-11-27:
## Environment
OS: Ubuntu 22.04
Qucs-s 24.3.1

## Expected Behavior
Primitive library devices have visible parameters that are updated in real-time (using callbacks) based on `w/l/other?` input settings

## Actual Behavior
`rfcmim` and `res` devices do not have `capacitance` nor `resistance` parameters, only `w/l`

Ikram-rs22, 2024-12-24:
How to calculate the values of resistance and capacitance from w & l? As far as I know, the sheet resistance of rppd is given between 235-285. Then how would yoi know the actual resistance of rppd? Same with cap_rfcmim.

sergeiandreyev, 2025-01-02:
Hi @Ikram-rs22, you can check the formulae to calculate resistance for `rppd` device in the TCL code of xschem symbol:
https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.tech/xschem/sg13g2_pr/rppd.sym
The same for `rfcmim` - these should be a part of device callback code.

sergeiandreyev, 2026-10-16:
Hi @abdur-rahman-dev, could you please check this issue if it's already resolved with new code and tools versions?

abdur-rahman-dev, 2026-10-19:
I have checked the current dev branch. all capacitors (except moscap of course) and resistors have capacitance and resistance value that updated with w/l.

closed 2026-10-19 by abdur-rahman-dev