IHP-GmbH/IHP-Open-PDK #285: LVS seems to have problems to recognize RPPD in .GDS

olisnr, 2024-12-02:
Environment
dev & main branch LVS scripts KLayout 0.29.8 linuX

Expected Behavior
LVS should see 5 RPPDs

Actual Behavior
it sees only 3

Steps to Reproduce the Problem
open GDS
select CLD
run LVS

[lvs_rppd.zip](https://github.com/user-attachments/files/17976498/lvs_rppd.zip)

sergeiandreyev, 2024-12-09:
Hi @olisnr, thank you for finding and reporting!
Are you running KLayout LVS in GUI or in batch mode?
When I'm using GUI I have the following error:
```
ERROR: In /home/andreev/opensource/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/macros/sg13g2_lvs.lylvs: Can't find a schematic counterpart for the top cell shunt_3V1 - use 'same_circuit' to establish a correspondence
ERROR: :/built-in-macros/_lvs_netter.rb:302: RuntimeError: Can't find a schematic counterpart for the top cell shunt_3V1 - use 'same_circuit' to establish a correspondence in Executable::execute
  :/built-in-macros/_lvs_netter.rb:302:in `align'
  (eval):2:in `align'
  /home/andreev/opensource/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/macros/../lvs/sg13g2.lvs:348:in `execute'
  :/built-in-macros/lvs_interpreters.lym:31:in `instance_eval'
  :/built-in-macros/lvs_interpreters.lym:31:in `execute'
  ```
  Is it the same error that you find on your side?

sergeiandreyev, 2024-12-09:
Hi @atorkmabrains, @FaragElsayed2, could you please take a look at this issue?

FaragElsayed2, 2024-12-10:
@sergeiandreyev 

For this error
```
ERROR: In /home/andreev/opensource/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/macros/sg13g2_lvs.lylvs: Can't find a schematic counterpart for the top cell shunt_3V1 - use 'same_circuit' to establish a correspondence
```

The issue arises because the top subcircuit name in the netlist doesn't match the layout name. You can either rename the netlist's top cell from `shunt_3V1_min` to `shunt_3V1`, or set `shunt_3V1_min` as the Top Cell when running the LVS.

@olisnr

The other resistors are not being recognized because they are connected using the GatePoly layer instead of Metal1. This alters the resistor structure, preventing it from being extracted as expected. I’ve updated the resistor extractor to use edges as ports rather than the full GatePoly port to accommodate such cases.

sergeiandreyev, 2024-12-10:
@olisnr, no, what is not matching in `pnpMPA`?

olisnr, 2024-12-10:
[image]

and also if i delete all the other things:
[lvs_pnpMPA.zip](https://github.com/user-attachments/files/18081185/lvs_pnpMPA.zip)

FaragElsayed2, 2024-12-10:
@olisnr Please check here for pnpMPA supported parameters
https://ihp-open-pdk-docs.readthedocs.io/en/latest/verification/lvs/04_03_bjt.html#pnpmpa

You need to update your cdl

```
Q3 VSS VSS net4 pnpMPA A=1.4p P=5.4u NE=1 M=1
```

sergeiandreyev, 2024-12-10:
but from KLayout is extracted like this:
```
.SUBCKT shunt_3V1 VSS
Q$1 VSS VSS \$10 pnpMPA AE=1.4p PE=5.4u AB=7.194p PB=10.96u AC=7.194p PC=10.96u
+ NE=1 m=1
.ENDS shunt_3V1
```

FaragElsayed2, 2024-12-10:
@sergeiandreyev 

We customized the supported classes in KLayout to map the parameters used (which are identical to parameters used in the commercial CDF) to their corresponding internal ones.

Explained in docs
[image]

sergeiandreyev, 2024-12-11:
@olisnr, how did you generate the CDL netlist? there were updates for the `pnpMPA` xschem symbol recently, so it has area and perimeter extracted:
[image]
[image]

olisnr, 2024-12-11:
@sergeiandreyev i used:
```
tclcommand="
	xschem set netlist_type spice
	set lvs_ignore 1
	set lvs_netlist 1
	set spiceprefix 0
	set last_local_netlist_dir $local_netlist_dir
	set local_netlist_dir 0
	xschem netlist [xschem get current_name].cdl
	set local_netlist_dir $last_local_netlist_dir
"
```
from a launcher:
[image]

it produces:
`Q3 VSS VSS net4 pnpMPA a={ 1.0u * 2.0u } p={ ( 1.0u + 2.0u ) * 2 }`

i hope i didnt missed a file: 
[shunt.zip](https://github.com/user-attachments/files/18094874/shunt.zip)

olisnr, 2024-12-11:
why arent the connections of the pnpMPA cell PLUS, MINUS & TIE instead of E B C?

KrzysztofHerman, 2024-12-18:
@olisnr since the 1RPPD1 problem is fixed we will close this issue. The discussion about `pnpMPA` will be continued in #284

PR #293 (2024-12-17): Improving resistor connections for LVS for proper detection with poly conn
### Improving resistor connections for LVS for proper detection with poly conn

Fixes #285
closed 2024-12-18 by KrzysztofHerman