IHP-GmbH/IHP-Open-PDK #286: PCELL for npn13G2 seems broken

olisnr, 2024-12-02:
Environment
dev & main branch LVS scripts KLayout 0.29.8 linuX

Expected Behavior
the PCELL should generate a **npn13G2** like in the example of 
`ihp-sg13g2/libs.ref/sg13g2_pr/gds/sg13g2_pr.gds
`
Actual Be (bottom left)
[image]
havior

Steps to Reproduce the Problem
place a npn13G2 from SG13_dev or main ...

KrzysztofHerman, 2024-12-02:
@olisnr 
It works but unfortunately you have to generate it in two steps. It is not documented so our bad. 
In general, in order to generate npn13G2 device you have to do it executing a particular sequence. 

1. Drag and drop a `npn13G2_base` device
2. Drag and drop a `npn13G2` device
3. Remove `npn13G2_base` device
4. Edit your cell.
5. If you need you can add more `npn13G2` devices

sergeiandreyev, 2024-12-09:
Hi @olisnr, can we close this one or is there anything else we can do for this issue? btw, I'm considering aligning different ways of instantiation between the `npn13G2*` devices, but it is not of a highest priority right now

olisnr, 2024-12-10:
somewhere it should be explained.

closed 2024-12-10 by olisnr