IHP-GmbH/IHP-Open-PDK #289: PCELL npn13G2 makes Min. nSD:block space to Cont (Note 2) = 0.00

olisnr, 2024-12-04:
## Environment
KLayout .29.8 mint/ubuntu 22.04 dev and main branch

## Expected Behavior
the PCELL npn13G2 should pass the DRC maximal script

## Actual Behavior
it produces
[image]

## Steps to Reproduce the Problem

1. make a npn13G2 from the PDK via instantiate npn13G_base the npn13G2
2. updte nx of the npn13G2
3. delete npn13G_base
4. run the maximum DRC script

sergeiandreyev, 2024-12-04:
Hi @olisnr, I guess this issue is related to how the DRC deck processes the hierarchy.
If you flatten the Pycell (Edit->Selection->Flatten Instances), you can see that the error is gone.. The `npn13g2` in `sg13g2_pr` GDS is flattened this is why it's clean (and, btw, the lower Metal1 drawing and pin layers are not aligned - we should fix this)
@akrinke, could you please take a look?

akrinke, 2024-12-04:
This is strange behavior. It seems I have a problem with my OpenPDK installation. Most of the time, I get the following error:
```
Warning: Dlo.exists: cell 'npn13G2_base' don't exists in library 'SG13_dev'!
(OA) Design "SG13_dev/npn13G2_base/layout" was not found
```
In my case, `nSDB.e` errors occur for both, the hierarchical and the flattened PyCell. It seems to be caused by a missing label `E` on the emitter Metal2 pin. Can you check if the label exists  in your tests?

olisnr, 2024-12-06:
@sergeiandreyev flatten dont works for me. the errors stay

sergeiandreyev, 2024-12-06:
yes, confirm, I've overlooked it 🤦‍♂️
I'm now checking on internal PDK if I have the same issue..

sergeiandreyev, 2024-12-06:
in commercial PDK the ref device layout is clean, it could be that we have some type of exceptions for these devices (I'm checking w/ the experts), also I'll double check with the layout from the open source

sergeiandreyev, 2024-12-06:
found strange behavior in KLayout - when I'm running DRC on the `npn13G2` from the `sg13g2_pr` library it's clean (what @olisnr was already mentioning), but if I go lower in hierarchy (and we have  here a leaf cell `npn13G2_base_CDNS_...`) I'm getting this `nSDB.e` violation
[image]
Is this expected?

sergeiandreyev, 2024-12-06:
btw, @akrinke, what do you think if it is feasible to provide somehow the actual notes from the DRM in the Marker Browser GUI, i.e. instead of 'Note 2' - 'nSD:block and Cont do not overlap'
It will be more evident for the user, and he will not be forced to open the DRM document to check what is written in the note..

akrinke, 2024-12-07:
@sergeiandreyev Yes, I will replace references to footnotes.

Here is what I've found: Running DRC on `npn13G2` from the `sg13g2_pr` library shows that it's clean (like both of you said). Now, removing the label `E` on the emitter Metal2 pin results in an `nSDB.e` error. This is also the reason why the leaf cell `npn13G2_base_CDNS_...` violates `nSDB.e`: it's missing the label `E` (and some other layers)! So this is expected.

My next test was to perform DRC on PyCells, flattened PyCells, and flattened PyCells with added label `E`. What is important: The layers `Activ.mask`, `TRANS` and `EmWind` are marked invalid and cannot be selected and copied! So I had to mark them as valid to make the copies. Especially the layer `TRANS` is important for `nSDB.e`. The result was, that adding the label `E` fixes the DRC errors.

[image]

akrinke, 2024-12-07:
@sergeiandreyev By the way, I've told you about problems creating `npn13G2` with `nx=2`. As you can see from the screenshot, these problems have disappeared. I don't know what caused them.

sergeiandreyev, 2024-12-07:
I'll update the Pycell to add labels

sergeiandreyev, 2024-12-09:
Hi @olisnr, @akrinke, could you please review the latest update in `dev` branch?

olisnr, 2024-12-10:
works for me. 
may be it should be written somewhere in the documentation for analog design?

akrinke, 2024-12-10:
Works for me too.

olisnr, 2024-12-11:
@sergeiandreyev yes. people need the instructions

closed 2024-12-18 by KrzysztofHerman