IHP-GmbH/IHP-Open-PDK #290: KLayout PCells Issue

mole99, 2024-12-06:
When I use the KLayout PCells, I get the following error:

```
ERROR: /home/leo/projects/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/pypreprocessor/pypreprocessor/__init__.py:11: ModuleNotFoundError: No module named 'imp'
```

My Python version is 3.12.7 and it seems that `imp` has been deprecated in favor of `importlib` in version 3.12.

A quick solution is to change the import in pypreprocessors `__init__` to:

```
import importlib as imp
```

A better solution would be to replace all occurrences of `imp` with `importlib`.

There's a commit in your pypreprocessor fork that added the `imp` library, I'm not sure if this is for a reason: https://github.com/IHP-GmbH/pypreprocessor/commit/d6088b2b7c8a91ba159b6735b8a1cabcb5039d90

akrinke, 2024-12-06:
Yes, I added the import because the file used `imp.lock_held()`. However, I got the same error a few days and removing `import imp` fixed it. Although I am not sure why, I think we can remove the import?

akrinke, 2024-12-06:
I guess, the line `imp.lock_held()` wasn't executed in my case. So your suggestion might be better.

mole99, 2024-12-06:
Good point! It seems that section isn't executed at all in our case. I can't really tell what its use is, you could probably try asking in the upstream repository, but it seems that it is no longer maintained as the last change was 6 years ago.

sergeiandreyev, 2024-12-08:
@ThomasZecha, could you please take a look at this issue?

adrienluitot, 2024-12-09:
Related to #187, might have to close both issues when the problem is fixed.

ThomasZecha, 2024-12-09:
I'll will take a look.

mole99, 2024-12-09:
Just checked and it works now as expected. Thanks!