IHP-GmbH/IHP-Open-PDK #296: Discrepancy in Bulk Substrate Thickness

Iranildot, 2024-12-16:
## Context
I was writing a .tek file for ASITIC app and I found the bulk substrate thickness measurement with two different values. 

- SG13G2_os_process_spec.pdf: 750 um
- Using OpenEMS Python with IHP SG13G2 v1.1.pdf: 750

## Screenshots

[image]
[image]

## Question

Which value is the right value?

KrzysztofHerman, 2024-12-16:
@Iranildot the valid answer is `280 um` and it should be used in your modeling. The `750 um` comes form the wafer processed in our clean room which is then mechanically postprocess to get different thicknesses.

olisnr, 2024-12-17:
@Iranildot You write a tech file for ASiTiC?

hpretl, 2024-12-18:
@sergeiandreyev Yes, I think that would be helpful.

KrzysztofHerman, 2024-12-18:
More information about the options of the wafer thickness offered by IHP can be found [here](https://www.ihp-microelectronics.com/services/research-and-prototyping-service/mpw-prototyping-service/schedule-price-list)

closed 2024-12-18 by KrzysztofHerman

VolkerMuehlhaus, 2024-12-20:
One more comment: the 280 micron bulk silicon thickness results in 300 micron total thickness of the stackup. Thickness specified by IHP is total chip thickness.