IHP-GmbH/IHP-Open-PDK #298: Inconsistencies in definitions of Rsil, Rppd, and Rhigh

RTimothyEdwards, 2024-12-17:
The definitions of resistor types Rsil, Rppd, and Rhigh differ between the DRC documentation, the primitive device library (which I assume comes from the klayout pycells) and the device DRC QA layouts.

Rsil:  The primitive device/pycell contains PolyRes which does not exist for this device in the documentation or the QA layout.

Rppd:  The primitive device/pycell contains PolyRes which does not exist for this device in the documentation or the QA layout.  Also, the primitive device/pycell does _not_ contain GatPoly in the resistor region, while both the documentation and QA layout have GatPoly.

Rhigh:  Both the primitive device/pycell contain PolyRes, but this layer does not appear in the device recognition definition in the documentation.

Please clarify how these devices are intended to be constructed.

sergeiandreyev, 2025-04-09:
Hi @stefansimon42, could you please take a look?

stefansimon42, 2025-04-10:
Hi @RTimothyEdwards 

Both GatPoly and PolyRes are physically the same polysilicon. Both layers are used as input for the polysilicon litho mask.

The difference is purely virtual. While GatPoly is defined as electrically conductive in the proprietary LVS, PolyRes is defined as non-conductive. On the one hand, this is intended to simplify the extraction of the devices, on the other hand, it prevents “virtual” shorts being generated in the LVS.

I will update the device description in the DRM. Thanks for pointing this out.

sergeiandreyev, 2025-04-10:
to add to this, indeed QA cells layouts should be updated, we have an internal discussion

RTimothyEdwards, 2025-04-11:
@stefansimon42 :  I think the discussion answers all of my questions, which I would summarize as follows:

1. PolyRes is intended to be mask-equivalent to GatPoly and is not simply an identifier layer.
2. The QA cells are incomplete/incorrect and will be updated.
3. The DRM description is incomplete/incorrect and will be updated
4. The primitive devices/pycells are correct and can be trusted

Thank you and @sergeiandreyev for the helpful answers.

olisnr, 2025-04-30:
@sergeiandreyev i have the question if its intended that the Rsil's PCell dont has a parameter "Bends"?

sergeiandreyev, 2025-04-30:
Hi @olisnr , yes, we removed it based on the feedback from the technology team and internal review

olisnr, 2025-05-02:
@sergeiandreyev parameter `b` should then be removed also from the` ihp-sg13g2/libs.tech/xschem/sg13g2_pr/rsil.sym` and may be from a QUCS symbol.

sergeiandreyev, 2025-05-02:
> [@sergeiandreyev](https://github.com/sergeiandreyev) parameter `b` should then be removed also from the` ihp-sg13g2/libs.tech/xschem/sg13g2_pr/rsil.sym` and may be from a QUCS symbol.

Hi @KrzysztofHerman, could you please take care?