IHP-GmbH/IHP-Open-PDK #300: QA cells: misunderstanding in Gat.b1 rule

mole99, 2024-12-18:
The GatPoly Gat.b1 rule is defined as follows:

| Rule   | Description                                                           | Value |
|--------|-----------------------------------------------------------------------|-------|
| Gat.b1 | Min. space between unrelated 3.3 V **GatPoly** over **Activ** regions | 0.25  |

The corresponding image is as follows:

[image]

To me, this would indicate that the spacing or notch of *related* HV GatPoly on the same Activ region is still defined by the Gat.b rule as 0.18um. Only for the HV GatPoly of unrelated regions of Activ the spacing is 0.25.

And yet the layout in the QA cells for Gat.b1 tests for the spacing/notch of HV GatPoly on a related region of Activ to be >= 0.25um.

| Pass  | Fail  |
|---|---|
| [image]  | [image]  |

sergeiandreyev, 2024-12-20:
Hi @mole99, forwarded to the internal team, but the reply for sure will be delayed after Christmas :-)

mole99, 2024-12-20:
Thank you, @sergeiandreyev! Yes, I didn't expect it any differently ;)