IHP-GmbH/IHP-Open-PDK #303: Incorrect illustration in DRC documentation

RTimothyEdwards, 2024-12-19:
In the PDF file `SG13G2_os_layout_rules.pdf`, section 5.11 "nSD:block" contains illustration Figure 5.11 "nSD:block dimensions".  However, the illustration is a duplicate of Figure 5.4 "nBuLay:block dimensions".  The layers and dimensions marked all correspond to nBuLay:block, not nSD:block.

sergeiandreyev, 2024-12-20:
Hi @RTimothyEdwards, this should be fixed in DRM v0.4 (https://github.com/IHP-GmbH/IHP-Open-PDK/commit/e261c6d59239b0afa7ca559aebb9dd6a163899c9)

sergeiandreyev, 2025-03-17:
Hi @RTimothyEdwards, gentle reminder to double check if the issue is fixed and we close it..

RTimothyEdwards, 2025-03-17:
@sergeiandreyev :  Yes, this has been fixed and I will close the issue.  Thanks!

closed 2025-03-17 by RTimothyEdwards