IHP-GmbH/IHP-Open-PDK #305: KLAYOUT Not Integrating with IHP Open PDK [ LIBRARY AND DEVICES ARE NOT FOUND]

Ikram-rs22, 2024-12-21:
I'm trying to integrate Klayout with the ihp open sg13g2 pdk but there seems to be a problem. I have integrated the pdk tech file with Klayout as shown below 

[image]

After that I exit and re-enter klayout but then this error is shown in the command line

[image]

And also there is no SG13_dev library in the library sections as a result no devices are found

[image]

But when I ran the pycell_test.py file a sg13_dev.gds file is created where I can see the devices

[image]

But still there is no library from which I can instantiate the cells. DRC and LVS options are visible and running but no devices. Please help.

adrienluitot, 2024-12-22:
Did you fetch the submodules? Like pypreprocessor and pycell4klayout-api

Ikram-rs22, 2024-12-23:
No I didn't. Where can I get those? @adrienluitot

Ikram-rs22, 2024-12-23:
Thanks the problem is solved. But I've noticed that when you increase or decrease the width and lengths of inductors and capacitors, the effect takes place but the value of inductance, capacitance showing in the paramter window doesn't change. So what to make of it?

adrienluitot, 2024-12-23:
Nice! Well done!

For the inductance and resistance value, I don’t think it is working. I only gave a quickly look at the PCell code, but didn’t found anything that updates those values. But it is something a little complicated to compute I think, I’m not sure there is a perfect formula.

As far as I know for the moment, the best thing to do is to generate a layout in KLayout, save the GDS and simulate your inductor in OpenEMS.

I started a Google Sheet with some simulated values (around 2.45GHz), I can share it if you wish.

Ikram-rs22, 2024-12-23:
Thanks it'd be a tremendous help if you shared your file.

akrinke, 2024-12-23:
We are currently working on a more accurate inductance extraction. It taps into the LVS, exports the geometry of inductors and runs FastHenry. The calculated inductance values are then added to the extracted netlist. We hope to have it packaged at the end of January.

Ikram-rs22, 2024-12-23:
Will the january pdk then have inductors as  schematic components? For circuit design?

adrienluitot, 2024-12-23:
> Thanks it'd be a tremendous help if you shared your file.

https://docs.google.com/spreadsheets/d/1vt_hs9XaM-SnFTcuAcVK475XKEWRAJqNsY20x1bT3ns/
There you go.
It is more of a draft than anything else, I might simulate more inductors with time. Maybe I might try to do a linear regression if I have enough simulations.

adrienluitot, 2024-12-23:
> We (TU Dresden) are currently working on a more accurate inductance extraction. It taps into the LVS, exports the geometry of inductors and runs FastHenry. The calculated inductance values are then added to the extracted netlist. We hope to have it packaged in a fork for testing at the end of January.

Good news ! It’s forgot about LVS, I don’t know it’s managed for the moment… maybe juste waived?

Anyway, glad to see some people are working on the inductors!

akrinke, 2024-12-23:
I guess our approach shouldn't be the default mode. Ideally, the inductor PCells (schematic and layout) should update the inductance value based on the geometry parameters. Then LVS just checks the geometry. However, using FastHenry makes sense for hand-drawn inductors.

adrienluitot, 2024-12-23:
I think most PDK just give well tested non-editable inductors. It seems to be harder to estimate the value of an inductor than of a resistor. But maybe this PDK could use a “simple” inductor formula to approximate its value to be LVS compatible. Maybe they already do that, I’m not sure.

I don’t know about FastHenry, I will inquire.

Ikram-rs22, 2025-01-03:
@sergeiandreyev for RF designof high frequencies (beyond 100 GHz), there are no RF probe pad or DC pads in klayout. As such, my intuition tells me it isn't possible to design RF circuits. Could you elaborate on that?

sergeiandreyev, 2025-01-03:
@Ikram-rs22, you mean the lack of `pad` device as KLayout parameterized cell (Pycell)? because we have `bondpad` layout in `sg13g2_pr` library
on the `pad` support - we're going to provide a temporary Pycell for KLayout in a short term, currently we have internal model update/development for this device in progress, but the expected time frame is somewhere in spring..

Ikram-rs22, 2025-01-03:
Did you mean this bondpad and cupillar pad? 

[image]

As far as I know any RF layout has pads for signal input and also dc biasing? will these pads suffice for those? And also for Dc - 150 GHz range? Moreover, is there any openMPW in March and will the pycell be available before that? Thanks in advance.

sergeiandreyev, 2025-01-06:
@Ikram-rs22, yes, I mean these cells
and yes, there will be 140 mm<sup>2</sup> available on March TapeOut for the open source community:
https://ihp-open-ip.readthedocs.io/en/latest/
and the plans are to provide the `pad` Pycell before that
- to your specific questions on pads I'll double check internally and come back to you

Ikram-rs22, 2025-01-13:
@sergeiandreyev any update on the RF GSG pads for RF signal path and DC pads for biasing? The rf_cmim cpacitor pycell is not present in layout but the qucs-s library has one? How to resolve this issue in layout where there is no pycell for rf_cmim??

sergeiandreyev, 2025-01-13:
Hi @Ikram-rs22, on the pads no updates for now, I'll have a discussion this week
On the `rfcmim` Pycell - it's in progress, should be ready this week or beginning next week worst case

Ikram-rs22, 2025-01-13:
Thanks a lot. Could you also kindly take a look at issue no 325?

closed 2025-05-12 by Ikram-rs22

Ikram-rs22, 2025-05-20:
What I think, you need to copy that cell to the new cell's cell window and then instantiate

engrvip123, 2025-05-20:
Sorry but I didn't get your above instruction. I don't see any IHP library to select device to instantiate its layout as shared in screenshot.

Ikram-rs22, 2025-05-20:
Please provide the terminal screenshot, if there occurs an error, the terminal shows it. It would be easier to debug from there-on @engrvip123

sergeiandreyev, 2025-05-20:
@engrvip123, do you have any warning/error messages in the terminal window (assuming you're running `klayout` from a terminal)

engrvip123, 2025-05-20:
@Ikram-rs22 @sergeiandreyev Here is the error seen in terminal.

[image]

Ikram-rs22, 2025-05-20:
@engrvip123 you haven't downloaded the klayout apis in the klayout/python/pycells4klayout folder

engrvip123, 2025-05-20:
@Ikram-rs22 Thanks for pointing out the issue. Can you please elaborate where to download these api's from and where to look for this folder `pycells4klayout` ?

Ikram-rs22, 2025-05-20:
You can from these 2 repos - 

https://github.com/IHP-GmbH/pycell4klayout-api/tree/047f1928ac6749d438a500c4ee90f32c879d3f11

https://github.com/IHP-GmbH/pypreprocessor/tree/6ddd5783365718c41eb69bce4428358ace4d1fa2

sergeiandreyev, 2025-05-20:
@engrvip123, based on the warning messages. it seems there is a mess up with the directory structure:
the `$HOME/.klayout` folder should not have any content inside `python` directory
The Pcell library is taken from `KLAYOUT_HOME`:
`/home/andreev/.klayout:/home/andreev/opensource/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout`
from `IHP-Open-PDK` part of the path

engrvip123, 2025-05-20:
@sergeiandreyev I am not very familiar with Linux or environment setup. Have followed steps from this link to install KLayout with the IHP PDK: [https://analog-course.readthedocs.io/en/latest/design_softwares/klayout.html.](url)

Below are the current settings in .bashrc used for Klayout with IHP pdk. Could you please review them and suggest any corrections to help resolve the error I'm encountering?

```
export PDK_ROOT=$HOME/IHP-Open-PDK
export PDK=ihp-sg13g2

export KLAYOUT_PATH="$HOME/.klayout:$PDK_ROOT/$PDK/libs.tech/klayout"
export KLAYOUT_HOME=$HOME/.klayout
export PATH="$PATH:$HOME/src/tools/klayout"
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$HOME/src/tools/klayout"
```

Ikram-rs22, 2025-05-20:
Your klayout path is wrong. Follow @sergeiandreyev's instruction

Ikram-rs22, 2025-05-20:
@engrvip123 you could follow this issue -
 https://github.com/IHP-GmbH/IHP-Open-PDK/issues/219

engrvip123, 2025-05-20:
> [@engrvip123](https://github.com/engrvip123), based on the warning messages. it seems there is a mess up with the directory structure: the `$HOME/.klayout` folder should not have any content inside [...]

@sergeiandreyev I have updated `KLAYOUT_HOME` variable and made `$HOME/.klayout/python` folder empty as suggested but issue still persists. Should I uninstall Klayout and install again? If yes, please suggest what steps to follow for new installation.

```
export PATH="$PATH:$HOME/src/tools/klayout"
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$HOME/src/tools/klayout"

export PDK_ROOT=$HOME/IHP-Open-PDK
export PDK=ihp-sg13g2 
export KLAYOUT_PATH="/home/$USER/.klayout:$PDK_ROOT/$PDK/libs.tech/klayout"
export KLAYOUT_HOME=/home/$USER/.klayout:/home/$USER/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout
```

[image]

[image]

engrvip123, 2025-05-20:
I checked and `.../IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/pycell4klayout-api` was empty. I cloned IHP pdk again as per instructions in above shared link. Now while invoking klayout getting following errors in terminal. 

[image]

sergeiandreyev, 2025-05-20:
if you used `git checkout dev` and it doesn't populate the sub-modules from the `dev` branch - we should change the documentation

engrvip123, 2025-05-21:
@sergeiandreyev I am using `dev` branch only of IHP pdk. 

[image]

mole99, 2025-05-21:
@engrvip123 can you verify that the active commit of `pypreprocessor` is `cf1ff9bad0fb5338cf1c5b990b2b816b1ea01a64`?

You may need to do another `git submodule update --recursive` after switching to the `dev` branch.

mole99, 2025-05-21:
And if you take a look at `IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/pypreprocessor/pypreprocessor/__init__.py` at line 11, does it say `import importlib` or `import imp`?

engrvip123, 2025-05-21:
I took help of chatgpt and issue got resolved now with klayout opening without errors and also I can see IHP pcell library. Here is the summary:

Issue:
KLayout threw a `ModuleNotFoundError: No module named 'imp'` when loading the IHP Open PDK because the PDK's PyCell scripts used the deprecated imp module, which is removed in `Python 3.12`.

Resolution:
Instead of downgrading Python, the scripts were updated to replace `imp.load_source()` with `importlib.util` and `exec_module()—`making them compatible with Python 3.12. After this change, KLayout ran without errors.

sergeiandreyev, 2025-05-22:
> I took help of chatgpt and issue got resolved now with klayout opening without errors and also I can see IHP pcell library. Here is the summary:
>
> Issue: KLayout threw a `ModuleNotFoundError: No [...]

please run `sed -n '11p' IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/pypreprocessor/pypreprocessor/__init__.py`
if you have:
`import importlib` - you have **correct** module version
if you have:
`import imp` - you have **incorrect** module version
instead of downgrading Python or updating scripts, just try to download the exact version which has to be used.
How to debug:
review the IHP OpenPDK GitHub repository through the web interface:
https://github.com/IHP-GmbH/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.tech/klayout/python
try to understand the Python error message and find the line with the code which is the root cause, compare the file where you have the error (`__init__.py`) with the file on the GitHub repo `dev` branch
try to learn/understand how to work in different git branches, download specific versions (commits), working with submodules...
P.S. IMHO, you should not expect ChatGPT to provide you the **optimal** answer, and I would strongly discourage to use it if you have very limited understanding of the topic you want it to help you with (what I mean, basically, you should have enough knowledge and be able to "trust" ChatGPT response/code/etc.)
and, BTW, it's not able to fully parse GitHub repositories

engrvip123, 2025-05-22:
Thanks @sergeiandreyev for your suggestion. I will definitely follow it. Following is output for `sed` command run at my end:
```
TT10:~$ sed -n '11p' IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/python/pypreprocessor/pypreprocessor/__init__.py
import importlib
```

But I am struggling with `ModuleNotFoundError: No module named 'klayout'` error while running LVS using `run_lvs.py` script. Can you please suggest about issue here?

sergeiandreyev, 2025-05-22:
@engrvip123, let's move step by step..
what did you had to do to have the `pypreprocessor` library now **not** referencing the `imp` module (which command did you use? did you manually edit the file?)
what was the root cause of the error?

engrvip123, 2025-05-23:
@sergeiandreyev  The error occurred because the pypreprocessor library was relying on the deprecated `imp` module, which was removed in Python 3.12. My system was running Python 3.12.3, leading to the `ModuleNotFoundError`.

I manually edited the file in the PDK that was importing` imp`. I replaced the deprecated `imp.load_source()` call with the modern equivalent using` importlib.util`.