IHP-GmbH/IHP-Open-PDK #309: Signal Amplitude Depends on Simulation Duration in Ngspice 43 Transient Simulation

rupokdas147, 2024-12-24:
## Environment
Ngspice 43
Qucs-s 24.3

I am encountering an unexpected behavior with Ngspice version 43 during the transient simulation. The issue is related to the amplitude of the signal, which seems to depend on the simulation duration. This problem does not occur in Ngspice version 42. Below is a detailed explanation:

Input signal: A squared current source:
Amplitude: 100 uA
Duration of High Pulse: 20 ps
Duration of Low Pulse: 20 ps
Rise Time: 20 ps
Fall Time: 20 ps

The screenshots attached with the description were the transient simulated output of an amplifier designed using an SG132 device.

## Expected Behavior
The expected behavior is created using **Ngpice 42**.

With 2 ns simulation duration, The signal amplitude:
[image]

With 3 ns simulation duration, The signal amplitude:
[image]

## Actual Behavior
The Actual behavior is created using **Ngspice 43.**

With a 2 ns simulation duration: The signal amplitude is very low:
[image]

Witha  3 ns simulation duration: The signal amplitude is very high:
[image]

## Steps to Reproduce the Problem
1. Create a simple circuit that generates a square wave signal with long rise and fall times. 
2. Run a transient simulation with different simulation durations (e.g., 2 ns and 3 ns).
3. Observe the output signal's amplitude.

KrzysztofHerman, 2024-12-28:
@rupokdas147  interesting case. As @StefanSchippers  wrote, please provide the testcase to reproduce the issue.

rupokdas147, 2024-12-28:
@StefanSchippers @KrzysztofHerman, I have changed the circuit a little bit. The output will not be the same as the previous screenshot, but it will be similar. 
[CE_Buffer_2_eye_diagram.zip](https://github.com/user-attachments/files/18265999/CE_Buffer_2_eye_diagram.zip)
I have tested for three different transient times: 2 ns, 3 ns, and 5 ns. For **Ngspice 43**, the amplitude was different for each case, but the amplitude remained the same for **Ngspice 42**.

StefanSchippers, 2024-12-28:
@rupokdas147 some comments:
the CE_Buffer_2_eye_diagram.cir defines a subcircuit with 6 ports (highlighted in bold):
.SUBCKT IHP_PDK_nonlinear_components_npn13G2  **gnd c b e bn t** Nx=1
...
.ENDS

but the instances call the subcircuit with 5 nets (in bold):
Xnpn13G1 **0  _net2 _net3 _net4 0** IHP_PDK_nonlinear_components_npn13G2 Nx=10

I have added a 0 (ground net) at the end of port list the 3 instance lines:

Xnpn13G1 0  _net2 _net3 _net4 0 **0** IHP_PDK_nonlinear_components_npn13G2 Nx=10

otherwise ngspice does not accept the circuit.

I have then run 4 simulations from 0 to 2ns with different timesteps: 1e-11, 1e-12, 1e-13, 1e-14
i get 4 waveforms that are completely different:

[image]

StefanSchippers, 2024-12-28:
Byincrementing the precision:
.option method=GEAR itl4=10000  sparse CHGTOL=1e-16 ABSTOL=1e-14 GMIN=1e-14 RELTOL=0.0001

I get this:

[image]

so spread is lower.

When such things happen either the circuit is unstable and has multiple operating points or it heavily depends on very small charges / currents. 
Increasing timestep does not produce better results if circuit depends heavily oncharge conservation. More timesteps means more accumulation of rounding errors.

StefanSchippers, 2024-12-28:
attached my modified .cir file
[CE_Buffer_2_eye_diagram_stef.zip](https://github.com/user-attachments/files/18268014/CE_Buffer_2_eye_diagram_stef.zip)

rupokdas147, 2024-12-29:
@StefanSchippers Please run by changing the duration of the simulation. 0 to 3 ns, 0 to 4 ns and 0 to 5 ns, and observe if there is any change in the output waveform or not.

rupokdas147, 2024-12-29:
@StefanSchippers @KrzysztofHerman

However, using Xyce as a simulator provides stable results.
[image]

[CE_Buffer_2_eye_diagram_xyce.zip](https://github.com/user-attachments/files/18268697/CE_Buffer_2_eye_diagram_xyce.zip)

rupokdas147, 2024-12-29:
According to my observations:
1. In **S-parameter** Simulation, **Xyce** and **Ngspice 43** provide almost the same result.
2. In **Transient** Simulation, **Xyce** and **Ngspice 42** provide almost the same result.

StefanSchippers, 2024-12-29:
@rupokdas147 I agree with you. Xyce runs provide identical results at any (reasonable) timestep given in the .tran comman. Moreover Xyce simulates this circuit orders of magnitude faster.

[image]

StefanSchippers, 2024-12-29:
And I also confirm that by rolling back to ngspice 42 simulation is aligned with Xyce and super fast. No special .option settings are required.

rupokdas147, 2024-12-29:
@StefanSchippers The **Noise Simulation** result is different in **Ngspice 42, 43 and Xyce**. At this point, I am really confused about my design. Should I go to layout or still there are problems with the schematics.

dwarning, 2025-01-14:
@StefanSchippers @rupokdas147 
Please substitute the attached library file sg13g2_hbt_mod.lib in the directory IHP-Open-PDK/ihp-sg13g2/libs.tech/ngspice/models. (Perhaps you make a backup before.)
[sg13g2_hbt_mod.zip](https://github.com/user-attachments/files/18404504/sg13g2_hbt_mod.zip)
The file has an additional switch for vbic's NQS effect: sw_nqs. By using an appropriate symbol the user can switch on/off this effect.
Please report if now the results from ngspice version 43 (and 44) fulfill the expectations. Then I will make a PR to IHP's dev branch.
Thank you.

rupokdas147, 2025-01-16:
@dwarning Thanks, **it worked**. I have tested using this file and now the Transient result is **not dependent on time step**.

StefanSchippers, 2025-01-16:
@dwarning sorry I can not simulate with the new .lib file, there are probably some implicit changes that must be done (5 terminal npn? 4 terminal npn ? don't know, ngspice reports "`Too many parameters for subcircuit type "npn13g2" (instance: xx1)`" , but since I don't know exactly the circuit I think @rupokdas147 result is the authoritative answer.

rupokdas147, 2025-01-16:
@StefanSchippers I have added `.INCLUDE path_to_PDK/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib` for each element in `IHP_PDK_nonlinear_components.lib` file. 

[IHP_PDK_nonlinear_components.zip](https://github.com/user-attachments/files/18438333/IHP_PDK_nonlinear_components.zip)

rupokdas147, 2025-01-16:
@StefanSchippers Here is my testcase. 

[CE_Buffer_3_eye_diagram.zip](https://github.com/user-attachments/files/18438381/CE_Buffer_3_eye_diagram.zip)

rupokdas147, 2025-01-16:
@StefanSchippers My PDK us from `dev` branch.

StefanSchippers, 2025-01-16:
Ah, ok, thank you!...

StefanSchippers, 2025-01-16:
@rupokdas147 sorry, but I really can't make it run and I think I throw the towel on this.
I have re-cloned the whole IHP pdk, switched to dev branch, put the new @dwarning file into my repo:
`.../IHP-Open-PDK_git/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib`:
```
schippes@asus:/mnt/sda7/IHP-Open-PDK_git$ git status
On branch dev
Your branch is up to date with 'origin/dev'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)

[CE_Buffer_3_eye_diagram_stefan.zip](https://github.com/user-attachments/files/18439479/CE_Buffer_3_eye_diagram_stefan.zip)

[CE_Buffer_3_eye_diagram_stefan.zip](https://github.com/user-attachments/files/18439487/CE_Buffer_3_eye_diagram_stefan.zip)

  (use "git restore <file>..." to discard changes in working directory)
	modified:   ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib

no changes added to commit (use "git add" and/or "git commit -a")
```
ngspice on your latest `CE_Buffer_3_eye_diagram.cir `shows the same message.
I think there are local changes on the PDK I don't have here.
```
schippes@asus:~/.xschem/xschem_library/CE_Buffer_2_eye_diagram$ ngspice -n -i CE_Buffer_3_eye_diagram.cir
******
** ngspice-44+ : Circuit level simulation program
** Compiled with KLU Direct Linear Solver
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2024, The ngspice team.
** Please get your ngspice manual from https://ngspice.sourceforge.io/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Mon Jan 13 16:37:42 UTC 2025
******

Note: No compatibility mode selected!

Circuit: * qucs 24.3.0  /home/rf/qucsworkspace/tia_rupok_prj/ce_buffer_3_eye_diagram.sch

Too many parameters for subcircuit type "npn13g2" (instance: xx1)
    Simulation interrupted due to error!
```

StefanSchippers, 2025-01-16:
The netlist I was trying to simulate.
the only changes are the paths of the .include and .lib lines.

[CE_Buffer_2_eye_diagram.zip](https://github.com/user-attachments/files/18439497/CE_Buffer_2_eye_diagram.zip)

StefanSchippers, 2025-01-16:
@rupokdas147 **update**: it is my ngspice version that fails. Rolled back to a ngspice-42 I have here and everything works now. Sorry for the trouble.

StefanSchippers, 2025-01-16:
Ok, thank you. the ngspice 44.2 and pre-master-45 fail.
ngspice-44 confirmed to work.
Thank you

closed 2025-01-16 by rupokdas147

dwarning, 2025-02-03:
@rupokdas147 
@StefanSchippers 

I have fixed the vbic model problem regarding simulations in time domain with nqs effect enabled (td > 0). The picture is made with ngspice 44+ (branch pre-master-45).
Output voltage is 60mVpp.

[image]

PR #331 (2025-01-22): provide a switch for vbic NQS effect
Fixes #309

The switch sw_nqs is a parameter of the npn13G2 subckt. Default is 0 - setting it to 1 enables the NQS model in vbic.
In schematic an appropriate symbol is needed.