IHP-GmbH/IHP-Open-PDK #310: Ngspice-32 or 43 Cannot calculate minimum noise figure (NFmin) if S2P file is used

Atif07Ali, 2024-12-27:
## Environment
Ngspice  42 or 43
Qucs-s 24.3
I am trying to design an LNA at 170 GHz frequency using SG13G2. I tried to build the circuit with transmission lines, using the S2P files of transmission lines. I faced the problem of 'nan' in the minimum noise figure (NFmin). It seems that the simulator cannot calculate NFmin.
if I use lumped elements, the NFmin gives the right value. However, when I use S2P files as the transmission lines, the NFmin does not provide any value. It seems that the simulator cannot calculate noise figure when there are S2P files. 

## Expected Behavior
The behavior has been simulated using lumped elements only. 
[image]

## Actual Behavior
This is the behavior after using S2P files.
[image]
[image]

## Steps to Reproduce the Problem

1. Create the circuit like the schematic.
2. Provide any S2P file to the 2-port s-param. 
3. Turn on the noise in "S parameter simulation" and simulate. plot nfmin.

dwarning, 2024-12-28:
I am afraid that noise contributions by S2P files can't simulate by ngspice, even if the files contain additional noise data.

But NaN looks more as bug. To investigate that it would be easier you could provide the schematic and the four S2P files. Or alternative the spice4cucs.cir file would be helpful too.

KrzysztofHerman, 2024-12-28:
@Atif07Ali @dwarning  indeed I have reproduced this testcase a few minutes ago and got the same  results, I mean `NaN`.
[Here](https://files.slack.com/files-pri/T01699QAZBQ-F086EG1DWG4/download/lna.rar?origin_team=T01699QAZBQ) is the complete testcase (one have to update the fixed paths pointing to the S2P files)

dwarning, 2024-12-28:
Slack channel is problematic - I believe I removed my membership.

Atif07Ali, 2024-12-29:
@dwarning thanks for your reply. I am sharing the test case and the 4 s2p files in zip format. The S2P file names are marked in red within the schematic. 
[LNA.zip](https://github.com/user-attachments/files/18268657/LNA.zip)

dwarning, 2024-12-30:
Thank you!
We have a problem in principle with simulating noise in s-parameter environment as we can see for resistors lower then system impedance:

[image]

After further investigations maybe we should open an ticket in ngspice forum.

dwarning, 2024-12-30:
It is pre-master-44 branch (not actual anymore because new version 44 just released): ngspice-43+.

Atif07Ali, 2024-12-31:
@dwarning There is a significant difference in results based on the version of ngspice. I am using both ngspice-42 and ngspice-43 for noise simulation. This is another problem. I think we should create an issue in ngspice forum. I am creating the issue. Please add your concerns in the reply.

dwarning, 2025-01-05:
After solving the NaN problem in ngspice coming back to the initial problem of the missing noise results in the LNA stage by using s-parameter files.

I can only recommend converting s-parameter files in general by using vector fitting with the scikit-rf python package: [scikit-rf](https://github.com/scikit-rf/scikit-rf). This has the additional advantage that AC, DC, Transient and Noise analysis are feasible.
[image]

Attached are the used python files, converted spice files and the qucs-s schematic.

@Atif07Ali Could you please check the plausibility of the noise and other results. 

@KrzysztofHerman We should find a way to support the converting of s-parameter files in general in the design flow.

[LNA_spar_noise_prj.zip](https://github.com/user-attachments/files/18309872/LNA_spar_noise_prj.zip)

Atif07Ali, 2025-01-05:
@dwarning thanks. Let me check it.

Atif07Ali, 2025-01-05:
[image]
[image]
@dwarning was able able to reproduce the figure as per your suggestion. I have simulated it both ngspice-42 and 43 version. The 42 version gives the nan value.

Atif07Ali, 2025-01-05:
[image]
[image]
Here if you see the nfmin values, there is a huge difference. ngspice-42 gives more realistic values (nfmin=4.48 @170G). After implementing the matching circuit, ngspice version 42 gives nan value in some frequencies according to previous case (with the use of lumped elements, it also give nan value). But the version 43 does not have the problem with nan. But 43 gives impractical values. I am actually confused which version should I use?

dwarning, 2025-01-05:
You are mixing lot of things in your discussion which I am not able to follow.
Version 42 is not acceptable. We are now with 44.
Please reduce the diagrams.
Let us start with one kind of matching elements, then the other types.
Last schematic shows new elements, which are not available for me.
It's confusing, sorry.

Atif07Ali, 2025-01-05:
I am sorry. Maybe I got your point. I was confused with the version at this time. Your suggestion worked perfectly for simulating s2p files.
Then the question is which version should I use?  43 gives impractical noise figure (nfmin) values. Should I try for 44 then?

dwarning, 2025-01-05:
Try to install 44 - needs about 5-10 min. Then we are on same level.
The vector fit approach is new. We have to investigate the noise performance of the generated networks.
Your help is appreciated.
Thanks!

engrvip123, 2025-12-28:
@Atif07Ali @dwarning
I would like to know whether the `nf` and `nfmin` parameters generated after enabling noise simulation in S‑parameter analysis in Qucs‑S represent absolute values of NF and NFmin, or if these values are expressed in dB.

Atif07Ali, 2025-12-28:
> @Atif07Ali @dwarning
> I would like to know whether the `nf` and `nfmin` parameters generated after enabling noise simulation in S‑parameter analysis in Qucs‑S represent absolute values of NF and NFmin, or if these values are expressed in dB. If it is absolute, how to generate and plot it in dB?
> 

Ngspice gives result in dB scale