IHP-GmbH/IHP-Open-PDK #313: False Seal.d errors

dnltz, 2024-12-31:
My design has a 2014.74 (w) x 2015.04 (h) sealring and reports all 7 violations for `Seal.d`. I quickly checked the sealring but wasn't able to verify something wrong. 

[image]

I was able to easily reproduce the issue: Check out latest dev branch (e261c6d59239b0afa7ca559aebb9dd6a163899c9), open Klayout and create a new layout. Place the default sealring at 0x0 and run `sg13g2_maximal.drc`. It should report the same errors (and obviously even more).

dnltz, 2024-12-31:
These rules were introduced with: https://github.com/IHP-GmbH/IHP-Open-PDK/commit/795e23ee3f4d7b4ea244574a1a780875c9a87974

So, question is whether the current sealring is wrong or those rules are not implemented properly.

sergeiandreyev, 2025-01-02:
@dnltz, thank you for reporting!
Hi @akrinke, could you please take a look?

akrinke, 2025-01-07:
Thanks for the report! The PR mentioned above should fix this issue.

sergeiandreyev, 2025-01-08:
Hi @dnltz, could you please check that the latest update fixes the issue on your side as well?

dnltz, 2025-01-08:
> Hi @dnltz, could you please check that the latest update fixes the issue on your side as well?

Yes, will do that this week. Thanks for the fix @akrinke

dnltz, 2025-01-09:
Just checked this fix and my layout is DRC clean again. Thanks!

PR #319 (2025-01-08): Update KLayout DRC scripts
- fix overlapping check used for example by `Seal.d` checks

Fixes #313
closed 2025-01-09 by dnltz