IHP-GmbH/IHP-Open-PDK #315: Undefined Parameter vbic_cje in NGSpice Netlist Simulation with SG13G2 Model Library

ArashKat, 2025-01-03:
Running the following netlist in ngspice:

* Example NGSpice netlist
* Circuit description
V1 net2 GND DC 3       * Voltage source with 3V DC
R1 net2 net1 1k m=1    * Resistor R1 with value 1k ohm
R2 net1 GND 1k m=1     * Resistor R2 with value 1k ohm
R3 net3 GND 1k m=1     * Resistor R3 with value 1k ohm

* Include the transistor model library
.include "/home/arashka/IHP-Open-PDK/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib"

* Define the transistor subcircuit instance
XQ1 net2 net1 net3 GND npn13G2 Nx=2

Netlist line no. 0:
Undefined parameter [**vbic_cje**]
Netlist line no. 0:
Cannot compute substitute

dwarning, 2025-01-05:
It seems there is something wrong with your input file.
Please attach the full netlist. 
BTW: inline comment is ";" - not "*" as shown above.

KrzysztofHerman, 2025-01-05:
@ArashKat 
please use `.lib cornerHBT.lib hbt_typ` instead of `.include`.

ArashKat, 2025-01-05:
Thank you; I made many changes even adjusting some variables inside the mod.lib based on information from cornerHBT, and it was working. Then I noticed I was running Xschem in the wrong folder after watching a tutorial on PDK importing another open-source PDK. My problem now is that after running on xschem I do not have the simulation result directly. I should go to the folder containing the spice netlist and run ngspice to see the results. Can you help me with that? Also, as I'm used to Cadence virtuoso, is there any way to import the pdk to Cadence,I tried some basic efforts such as adding pdk folder address to cds.lib in Cadence but I couldn't succeed....

KrzysztofHerman, 2025-01-05:
@ArashKat happy to hear that you have made some progress. 
The easiest way to get familiar with the tools and PDK is to fire up xschem in `libs.tech/xschem` directory and review the examples. By default xschem will create a subdirectory called simulations there and it will be used by ngspice as `cwd`

ArashKat, 2025-01-06:
Thank you. Now everything is working for me but I have three questions:

1- Is it possible to see the simulation result directly in Xschem?

2- Is it possible to use this pdk in Cadence 

3- What is the nominal voltage for VCC in this technology.? I see VDD is 1.2V. I saw the breakdown voltage in the documents, but not a basic definition, reference, or suggestion for VCC.

Thank you if you please respond to these questions.

KrzysztofHerman, 2025-01-06:
@ArashKat 

Great to know that you have managed to make things working.  As for your questions:

1. yes it is possible. You can plot the results using the graph element directly on the schematic. It sources the *.raw file generated by ngspice. Almost all our examples use this method. Alternatively you can use `GAW` waveform editor, which does the same standalone . 
2. No, the PDK is targeting open source EDA tools and this is the reason why it is incompatible with Cadence tools. Nevertheless the IHP-Open-PDK is derived from our proprietary product, compatible with Cadence, and if you are with an academic institution you can get access to it free of charge. The requirement is NDA.
3. I suppose you are asking about the standard cells and standard cells supply voltage. The nominal one is 1.2 V however it will work even with 1.5 V (we provide a separate corner liberty file which uses 1.5 V)

sergeiandreyev, 2025-01-07:
> 3- What is the nominal voltage for VCC in this technology.? I see VDD is 1.2V. I saw the breakdown voltage in the documents, but not a basic definition, reference, or suggestion for VCC.

Hi @ArashKat, if you're asking about high-voltage supply level - it's 3.3V

sergeiandreyev, 2025-01-31:
Hi @ArashKat, was the original issue (first description) resolved? If yes, we would like to close this case.
We can have separate discussion in separate 'issue' if you want.

ArashKat, 2025-01-31:
Hello @sergeiandreyev,

Yes, it is over. I will open another issue.

closed 2025-02-03 by sergeiandreyev