IHP-GmbH/IHP-Open-PDK #317: Ngspice cannot run transient with s2p files in matching networks

shafinhossain, 2025-01-07:
## Environment
qucs-s  24.3
ngspice 43

## Expected Behavior
[image]
Using Ideal Inductors in the matching network of the Power Amplifier (180 GHz) transient simulation and S parameter simulation both are successful.

## Actual Behavior
[image]
Using openEMS the ideal inductors were replaced by S2P files. But this time the simulator could not run Transient Simulation but does support S parameter simulation. 
Note: Only occurs when the inductors in the matching network are replaced by S2P files, however when the RF Choke or Degenerative Inductor is replaced by S2P files, transient does run.

## Steps to Reproduce the Problem

1. Create an amplifier circuit with npn13g2, and proper matching networks (L matching)
2. Replace the inductors in L matching with S2P file.
3. Run transient simulation and see it fail.

KrzysztofHerman, 2025-01-07:
@shafinhossain please se the #314 issue

sergeiandreyev, 2025-01-27:
Hi @shafinhossain, @KrzysztofHerman, so, should we close this issue as 'duplicate'?