IHP-GmbH/IHP-Open-PDK #322: Importing MOS library

ArashKat, 2025-01-08:
The following error exists in simulating MOSs:

**Error** on line:
  n.xm1.nsg13_hv_pmos net2 net1 0 0 xm1:sg13g2_hv_pmos_psp w=    1.000000000000000e-06     l=    4.500000000000000e-07     nf=    1.000000000000000e+00     mult=    1.000000000000000e+00     as=    3.400000000000000e-13     ad=    3.400000000000000e-13     ps=    2.680000000000000e-06     pd=    2.680000000000000e-06     dta=    0.000000000000000e+00     ngcon=2
could not find a valid modelname
    Simulation interrupted due to error!

**the netlist:**

**.subckt dc_hv_pmos
Vgs net1 GND -0.75
Vds net3 GND -1.5
XM1 net2 net1 GND GND sg13_hv_pmos w=1.0u l=0.45u ng=1 m=1
Vd net3 net2 0
.save i(vd)
**** begin user architecture code

.lib /home/arashka/IHP-Open-PDK/ihp-sg13g2/libs.tech/ngspice/models/cornerMOShv.lib mos_tt

.param temp=27
.control
save all
op
print I(Vd)
dc Vds 0 -2 -0.01 Vgs -0.45 -1.1 -0.05
write dc_hv_pmos.raw
.endc

**** end user architecture code
**.ends
.GLOBAL GND

KrzysztofHerman, 2025-01-09:
@ArashKat please verify your PDK installation.  Please checkout [this manual](https://ihp-open-pdk-docs.readthedocs.io/en/latest/analog/xschem.html). The core issue is probably:
1. you do not have the osdi model comiled
2. your `./home/$user/.spiceinit` file does not exists

Our MOS devices use PSP model written in Verilog-A, we distribute source code but the user have to compile it using open source tool openvaf. The binary have to be placed in `$PDK_ROOT/libs.tech/ngspice/openvaf`
and `.spiceinit` have to point to this location.

ArashKat, 2025-01-27:
Hello the manual was very helpful ... it solved the issue: however it doesn't recognize the resistors; Xschem crashes and suddenly will be closed when I select a pdk's resistor.

sergeiandreyev, 2025-01-29:
Hi @KrzysztofHerman, could you please take a look?

ArashKat, 2025-01-29:
For me, the problem is that after running the install.py I did not create any  r3_cmc file .... I added them manually using this link: https://github.com/dwarning/VA-Models/releases/tag/v2.2  by copying the VA-Models-osdilibs.zip file to $PDK_ROOT/$PDK/Lib.tech/ngspice/openvaf  directory and added the following command: osdi  '$PDK_ROOT/$PDK/libs.tech/ngspice/openvaf/osdilibs/r3_cmc.osdi' to .spiceinit in $HOME still Xschem jump out when I select the resistors

sergeiandreyev, 2025-03-17:
and please note that in the latest `dev` branch we changed the default location of osdi models to:
`osdi  '$PDK_ROOT/$PDK/libs.tech/ngspice/osdi/r3_cmc.osdi'`