IHP-GmbH/IHP-Open-PDK #323: Discrepency between Ngspice and Xyce simulation

adrienluitot, 2025-01-09:
## Environment
**OS:** Ubuntu  24.04
**Tools:** Qucs 24.4.1
**Simulator:** Ngspice-44  and  Xyce 7.8

## Expected Behavior
Having similar simulation results between Xyce and Ngspice.

## Actual Behavior
We are designing a simple LNA at 2.45GHz. Here is its schematic and some ngspice simulation results:
[image]
On the right we see the Re(Z11) is 47.047Ω (with Ngspice).

When I run the same design, with the same parameters/variables, and the same simulation, but with Xyce, I get these results:
[image]
The discrepancy is not negligible here, mostly for Re(Z11), almost 10% difference.

For other results, there is not such a discrepancy, for instance, gain is pretty similar: 15.531dB for Ngspice and 15.561dB for Xyce.

I don't know if the error comes from:
- Me (Xyce installation maybe ?)
- Ideal components (maybe Xyce and Ngspice treat R,L,C slightly differently -> I will soon try with IHP's real components)
- Discrepancy between IHP's Ngspice and Xyce models
- Something else

## Steps to Reproduce the Problem
1. Create a schematic in Qucs
2. Simulate with Ngspice
3. Copy whole schematic (design, parameters, simulations)
4. Adapt to Xyce (include & plugin)
5. Simulate with Xyce

KrzysztofHerman, 2025-01-09:
@adrienluitot thank you for reporting this issue. 
Since you use mosfets my first guess is that ngspice and Xyce use different Verilog-A compilers, `openvaf` and 'adms' respectively,  in order to make PSP binary. Moreover you operate at relatively high frequencies, where the `non quasi static` PSP model part  was disabled because of the `openvaf` compiler issues, what can bring us to this situation. If you wish you could investigate in this direction.

adrienluitot, 2025-01-10:
> @adrienluitot thank you for reporting this issue. Since you use mosfets my first guess is that ngspice and Xyce use different Verilog-A compilers, `openvaf` and 'adms' respectively, in order to make [...]

Thanks for your reply! Indeed that may be the compiler, but I believe that the compilers should follow the models? So that it should rely more on the models rather than the compilers?

I don't know much about `non quasi static`, can you tell me more about it?
It would a pleasure to investigate, but I don't really know where to go. If you have any direction, that would be great!

KrzysztofHerman, 2025-01-10:
@adrienluitot thank you for your interest.  If you wish you could simplify the circuit and lower the frequency by a decade and then see the results. This way we could discard `nqs`.  Our experience is that in general you will face differences between the simulation results using ngspice an Xyce.

adrienluitot, 2025-01-15:
Sorry for the late answer. So I used the schematic given above and run the simulation with Ngspice and Xyce around 2.45GHz, 245MHz and 24.5MHz. And here is what I had:

2.45GHz:
[image]
 
245MHz:
[image]

24.5MHz:
[image]

Sure the discrepancy is more present (in percentage at least) at high frequency, but it seems also present at lower frequencies.