IHP-GmbH/IHP-Open-PDK #324: GSG / RF Signal Pad and DC Pads Missing

Ikram-rs22, 2025-01-10:
For RF design from DC to W /D band , the NPN13G2 HBTs are excellent. But a predicament is posed as in the layout section there is no pycell for the DC bias pad and the RF signal pad. The RF input & output signal paths need GSG pads. Also, for RF design, Qucs-s provides cap_rfcmim schematic symbol but there is no pycell for it in klayout.

[image]

Although there are some bondpad and cupillar pad, these are not RF pads in the sg13g2_pr gds., The rf_cmim cell is also present here but there is no sg13g2_pr tech file for the library. So the pads are absent in both cases.

[image]

RF and DC pads are must for RF design for designs ranging from DC-140 GHz & beyond.

giove11, 2025-01-13:
Hi, about the pads, everyone design their own pads and simulated them, which is why there is no pcell for a GSG pads.
About the cap_rfcmim I don't know.

olisnr, 2025-01-13:
i would draw the PADs my self, and try to make transmission lines with the right impedance. PADs alone wont be help full. may be You have to look to Your RF-wafer probe for the dimensions. and simulate the impedances.

Ikram-rs22, 2025-01-13:
okay so what layers do I need for the pads? Transmission lines are a simple process but a GSG pad, especially if you're designing something from DC-150 GHz isn't an easy task to make. Would appreciate any further elaboration

KrzysztofHerman, 2025-01-13:
@Ikram-rs22 we have recently provided a pyCell for a bondpad so you can easily generate or walk through the code to figure out which layers you need.

shafinhossain, 2025-01-16:
@KrzysztofHerman We were actually willing to test our chips at IHP facility, and in order to do that we require certain information about the pads.
1. Pitch size (i.e. 30 um/50 um)
2. Dimension of the pads
(note: our designs have two frequency ranges, one being DC to 130 GHz and the G band (140-220 GHz)).
Additionally, if further communication with IHP personnel involved in testing is necessary, kindly let us know.

KrzysztofHerman, 2025-01-16:
@shafinhossain please refer to the `SG13G2_layout_rules.pdf` document You can find it in our PDK under `ihp-sg13g2/libs.doc/doc` directory. 

[image]

1. `Pad.bR` min. pad spacing is 8,4
2. it can be customized. In our dev branch you will find a dedicated pyCell called `bondpad` which implements it.

sergeiandreyev, 2025-01-17:
Hi @Ikram-rs22, the `rfcmim` Pycell is available in the `dev` branch, please check and give your feedback

Ikram-rs22, 2025-01-17:
@sergeiandreyev yes I've already checked it out. Also thanks for incorporating the capacitance value update feature with dimensional changes.

sergeiandreyev, 2025-01-29:
Hi @Ikram-rs22, so is there anything else you want to add to this issue, or we can close it?

Ikram-rs22, 2025-01-29:
You can close this issue. I'll have to come up with my own solutions. Thanks for your efforts.

closed 2025-01-29 by sergeiandreyev