IHP-GmbH/IHP-Open-PDK #325: Query about RF Layout in KLayout

Ikram-rs22, 2025-01-10:
I have some queries about RF layout in this pdk rule -

1) Like cadence, is the black portion of the screen considered as global p substrate?

[image]

2) About guard rings, is the HBT inside a guard ring? How do we create a guard ring for the resistors as the bulk connection is sub! ? The layers needed for creating a guard ring. Is the layers in the ptap pycell the ones for creating a guard cell?

[image]

3) Regarding the metal shielding in RF designs, is it a problem to have a global metal shielding for the routing paths and pycells or does it need to be chopped ? Is there any layer which will prevent the respective area where it is drawn from being covered in the global metal shielding meaning that part won't be counted in the shielding??

[image]

KrzysztofHerman, 2025-01-14:
@Ikram-rs22 
As for your questions:
1) SG13G2 is a bulk process and the wafer is `p-type` substrate so I can confirm that when you lay out your devices you effectively place it on the substrate. 
2)  We found out that there is a requirement from the designers to have such cell as guard ring. Please take a look at this [branch](https://github.com/KrzysztofHerman/IHP-Open-PDK/tree/ptap1-ntap1-fix) you can find there xschem symbols, ngspice models and klayout pycells of such devices. Soon it should be integrated in the dev branch. Please keep in mind that the guard ring is a modeled as a resistance between the particular net (usually GND) and the substrate. 
3) As for metal shielding you have a free hand here. Just keep in mind the slit rules. `"Large areas of metal are subject to mechanical stress during production. This can cause metal detachment from the oxide. The use of metal slits leads to reduction of mechanical stress."` (chapter 7.3 sg13g2_Layout_Rules manual)

Atif07Ali, 2025-01-15:
@KrzysztofHerman 
There is a layer in KLayout called NoRCX. For instance, if I place a metal2 layer on top of Metal1 without using any vias, should I use the NoRCX layer? I am designing an LNA at 170 GHz

sergeiandreyev, 2025-01-31:
Hi @Ikram-rs22, do you have more questions on the above items, or we can close the issue?

Ikram-rs22, 2025-01-31:
Please kindly close this issue

closed 2025-01-31 by sergeiandreyev