IHP-GmbH/IHP-Open-PDK #327: Unale to use PMOS with XYCE

EngGhaith, 2025-01-11:
## Environment
Xschem & XYCE

## Expected Behavior
Simulation runable

## Actual Behavior
XYCE Abort with this message:
_Netlist error in file
 /foss/pdks/sg13g2/libs.tech/xyce/models/sg13g2_moslv_mod.lib at or near line 86
 Unrecognized parameter D for device YPSP103_VA!M1
Netlist error in file
 /foss/pdks/sg13g2/libs.tech/xyce/models/sg13g2_moslv_mod.lib at or near line 86
 Unrecognized fields for device YPSP103_VA!M1
Netlist error in file
 /foss/pdks/sg13g2/libs.tech/xyce/models/sg13g2_moslv_mod.lib at or near line 86
 Unrecognized parameter D for device YPSP103_VA!M1
Netlist error in file
 /foss/pdks/sg13g2/libs.tech/xyce/models/sg13g2_moslv_mod.lib at or near line 86
 Unrecognized fields for device YPSP103_VA!M1
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 4 MSG_ERROR
 errors
*** Xyce Abort ***_

## Steps to Reproduce the Problem
[TEST_PMOS.zip](https://github.com/user-attachments/files/18386566/TEST_PMOS.zip)
Use this Nelist or simulate the Xschem-Schematic TEST_PMOS.sch:

`V1 net1 0 2.4
XM1 net2 net2 net1 net1 sg13_lv_pmos w=1.0u l=1.0u ng=1
XM2 net3 net2 net1 net1 sg13_lv_pmos w=1.0u l=1.0u ng=1
R1 net2 0 1k
R2 net3 0 1k
**** begin user architecture code

.TRAN 1p 20n
.PRINT TRAN FORMAT=RAW FILE=TEST_PMOS_TRAN.raw V(*) I(*)

.lib /foss/pdks/sg13g2/libs.tech/xyce/models/cornerMOSlv.lib mos_tt

**** end user architecture code
**.ends
.end`

EngGhaith, 2025-01-11:
NMOS seem to have same problem!

eskiyerli, 2025-01-11:
I think the problem is that the IHP has chosen to use Verilog-a models within Xyce models. Interestingly, Xyce distribution does not include a Verilog-A compiler. It can be only compiled on Linux as a plugin, but not on Windows. I was just scratching my head about the same problem. As it stands, IHP model libraries are not compatible with Xyce as distributed by Sandia.

EngGhaith, 2025-01-11:
@eskiyerli 
Iam not sure if that is the problem, because I am using Linux and MOSFETs do not work on Linux too.

eskiyerli, 2025-01-11:
@EngGhaith If you are using the Xyce binaries downloaded from Sandia, they don't include the Verilog-A plugin either.

EngGhaith, 2025-01-11:
@eskiyerli 
Do you have any manual to include this Plugin at Linux?

eskiyerli, 2025-01-11:
This is the PKGBUILD file I am using for Manjaro:
# Maintainer: Your Name <your.email@example.com>
pkgname=xyce
pkgver=7.8.0
pkgrel=1
pkgdesc="An open source, SPICE-compatible, high-performance analog circuit simulator"
arch=('x86_64')
url="https://xyce.sandia.gov/"
license=('GPL3')
depends=('trilinos-xyce' 'boost-libs' 'openmpi' 'fftw' 'suitesparse')
makedepends=('cmake' 'gcc' 'gcc-fortran' 'boost' 'python')
optdepends=('python: for Python scripting support')
source=("https://github.com/Xyce/Xyce/archive/Release-"$pkgver".tar.gz")
sha256sums=('f763b7d5ad6defd25d2c7e5cc95155958cd12510a5e22a179daab459b21fa713')  # Replace with actual SHA256 sum of the source tarball

build() {
  cd "${srcdir}/Xyce-Release-${pkgver}"
  mkdir -p build && cd build
  cmake  .. \
    -D CMAKE_INSTALL_PREFIX=/usr \
    -D CMAKE_BUILD_TYPE=Release \
    -D BUILD_SHARED_LIBS=ON \
    -D Xyce_PLUGIN_SUPPORT=ON
  make
}

package() {
  cd "${srcdir}/Xyce-Release-${pkgver}/build"
  make DESTDIR="${pkgdir}" install
  rm "${pkgdir}/usr/bin/admsXml"
}

eskiyerli, 2025-01-11:
There is also this information in adms folder about how to use this with xschem:
buildxyceplugin psp103.va .
cp Xyce_Plugin_PSP103_VA.so ../xschem/simulations/
mv Xyce_Plugin_PSP103_VA.so ../xschem/examples/

eskiyerli, 2025-01-11:
Finally success. Using make file in adms directory produces libXyce_Plugin_PSP103_VA.so linkable file. Copy it to same directory as your netlist file. For example, to examples directory. Then start Xyce with --plugin option:
Xyce -plugin ./libXyce_Plugin_PSP103_VA.so lv_nmos_id_vg_vb.cir

KrzysztofHerman, 2025-01-12:
@EngGhaith were you able to compile the shared object and run the simulation ? Indeed we use PSP model which should be compiled using sources distributed within the PDK. Due the `adms` and/or `openvaf` limitations  the sources may have modifications. Additionally due to the license issues we are not allowed to distribute the binary `.so` file so you have to compile it on your site.

EngGhaith, 2025-01-13:
I could use it in bare Xyce.
But I could not integrate it with Xschem.
I copied the Plugin: Xyce_Plugin_PSP103_VA.so to ihp-sg13g2/libs.tech/xschem/simulations/.
But it did not work.

KrzysztofHerman, 2025-01-13:
@EngGhaith in our dev branch you will find some examples where you can simulate either using ngspice or Xyce. 

[image]

The plugin is added in the `SimulateXyce` launcher 
```
set sim(spice,3,cmd) {Xyce -plugin $env(PDK_ROOT)/$env(PDK)/libs.tech/xyce/plugins/Xyce_Plugin_PSP103_VA.so \"$N\"}
```

EngGhaith, 2025-01-13:
Now it works!
Thank you very much!

KrzysztofHerman, 2025-01-13:
I will close as solved.

closed 2025-01-13 by KrzysztofHerman