IHP-GmbH/IHP-Open-PDK #328: Plotting MOS gm with Xyce

adrienluitot, 2025-01-13:
Hello everyone,

I tried to plot a MOS gm (and other variables) with Xyce (and Qucs-s).
In https://xyce.sandia.gov/files/xyce/Xyce_Reference_Guide_7.6.pdf I saw we can use `N(XMN0:gm)` in `.PRINT`. But it doesn't work, I get `Netlist error: Function or variable N(MN0:GM) is not defined`.

I think the syntax might be different because of ADMS?
Anyone knows the right syntax?

KrzysztofHerman, 2025-01-14:
@adrienluitot seems that the method you have found applies to BSIM3 and BSIM4 models. I have run a simple test which dumps all the PSP model and instance parameter. Additionally there is no `gm` parameter. I have also asked a question on the open source silicon slack channel. [Here](http://open-source-silicon.slack.com) you can join: 

```
** sch_path: /home/herman/tmp/xyce_op.sch
**.subckt xyce_op
Vin in GND pulse(0.0 1.2 0.0 100p 100p 2n 4n)
Vdd net1 GND 1.2
XM1 out in GND GND sg13_lv_nmos w=1.0u l=0.45u ng=1
XM2 out in net1 net1 sg13_lv_pmos w=1.0u l=0.45u ng=1
**** begin user architecture code
.preprocess replaceground true
.option temp=27
.op
.print tran format=raw file=tran_logic_not.raw
.lib /home/herman/github/KrzysztofHerman/IHP-Open-PDK/ihp-sg13g2/libs.tech/xyce/models/cornerMOSlv.lib mos_tt
**** end user architecture code
**.ends
.GLOBAL GND
.end

```

```
------------------------------------------------------------
Operating point information:
Number of ADMS PSP103_VA models: 2
          name XM1:SG13G2_LV_NMOS_PSPXM2:SG13G2_LV_PMOS_PSP
          type       PSP103_VA      PSP103_VA
         level               1              1
            DTA              0              0
         FJUNQD           0.03           0.03
       VJUNREFD            2.5            2.5
          FJUNQ           0.03           0.03
       SWJUNEXP              0              0
        PBRSTID              4              4
        PBRBOTD              4              4
        VBRGATD             10             10
        VBRSTID             10             10
        VBRBOTD             10             10
     STFBBTSTID         -0.001         -0.001
     STFBBTBOTD         -0.001         -0.001
      FBBTRSTID          1e+09          1e+09
      FBBTRBOTD          1e+09          1e+09
       CBBTSTID          1e-18          1e-18
       CBBTBOTD          1e-12          1e-12
    MEFFTATGATD           0.25           0.25
       CTATGATD         0.0001         0.0001
       CTATSTID         0.0001         0.0001
       XJUNGATD          1e-07          1e-07
       XJUNSTID          1e-07          1e-07
       CSRHGATD         0.0001         0.0001
     IDSATRGATD          1e-18          1e-18
       PHIGSTID           1.16           1.16
       PHIGBOTD           1.16           1.16
          PGATD            0.5            0.5
       VBIRGATD              1              1
       VBIRSTID              1              1
       CJORGATD          1e-09          1e-09
       CJORSTID          1e-09          1e-09
       CJORBOTD          0.001          0.001
         PBRGAT              4              4
         PBRSTI              4              4
       CBBTGATD          1e-18          1e-18
         PBRBOT              4              4
         VBRSTI             10             10
         VBRBOT             10             10
      STFBBTSTI         -0.001         -0.001
      STFBBTBOT         -0.001         -0.001
       FBBTRGAT          1e+09          1e+09
        CBBTBOT          1e-12          1e-12
[... 791 lines cut ...]
           FREQ              0          1e-10
            VHI              0              0
             Z0             50             50
             V1              0              0
             TR              0          1e-10
           TAU1              0          1e-10
             PW              0          2e-09
             V0              0              0
           TAU2              0          2e-09
             TD              0              0
            PER              0          4e-09
           DCV0            1.2              0
             RB              1              1
              V              0              0
            TD2              0          1e-10
           PORT              0              0
          ACMAG              0              0
             FC              0              0
              R              0              0
              T              0              0
            TD1              0              0
            VLO              0            1.2
        TSAMPLE              0          2e-09

```

adrienluitot, 2025-01-14:
Thanks for your help, indeed that what I though for BSIM3/4.
I saw there is no "Output Variables" for the PSP103 model in the reference guide. The report you sent seems to correspond to the " Instance Parameters", that might explain why there is no "gm"?

Also, I'm not sure the way I use the first part `N(XMN0...)` is the right way to target a PSP103 device. I see in your report that it might look to something closer to `XM1:SG13G2_LV_NMOS_PSP`.
How did you print the report you sent here? This might help me to understand.

KrzysztofHerman, 2025-01-14:
@adrienluitot
Let's see if someone answer the question on slack. There exists also Xyce google mailing group. You can ask there, I am curious as well. 
As for printing the values when I ran the test case in xschem a new window pops up, I have just save the results. 
I think, but I am not sure that it is possible to dump the results to a file using `-o` options in Xyce I mean to run 
```
Xyce -plugin $PDK_ROOT/$PDK/libs.tech/xyce/plugins/Xyce_Plugin_PSP103_VA.so -o results.txt  netlist.spice
```

adrienluitot, 2025-01-14:
I will ask on Xyce's GH, I had good answers last time. I don't have Slack, so I'll wait for you feedback on this side.

I'm dumb, sorry for the report question... I didn't check logs in Qucs-s when Xyce worked, and of course when Xyce gives an error there is no such report.

sergeiandreyev, 2025-01-14:
Hi @adrienluitot, just one remark..
please try to use recommended tool versions:
https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/versions.txt
https://xyce.sandia.gov/files/xyce/Xyce_Reference_Guide_7.8.pdf
It will be easier to align the analysis in case of differences between versions

adrienluitot, 2025-01-14:
Hello Sergei, I use Xyce 7.8, I just shared the first reference guide I found on the internet. I checked the 7.8 reference guide after, but there is no more information about the output variables.

tvrusso, 2025-01-14:
I had a stab at answering this question in Xyce/Xyce#127, but will give a quick summary of that answer here.

Xyce's `.op` is very seriously limited and doesn't output any "output" variables.  It does nothing but dump the input parameters of the models.  gm isn't an input parameter of the PSP, so that's why you don't see it in the `.op` output.

The other issue is that the X devices aren't actually MOSFET devices, they're subcircuit calls --- the actual mosfet devices are the Y devices inside the subcircuit, and so to print them with the N() syntax you have to provide the complete path to the actual device, not the subcircuit name alone.

In the case of the [netlist above](https://github.com/IHP-GmbH/IHP-Open-PDK/issues/328#issuecomment-2589169177), the XM1 subcircuit instance is an instance of the `sg13_lv_nmos` subcircuit defined [in the PDK](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/xyce/models/sg13g2_moslv_mod.lib).  That subcircuit instance contains a "YPSP103_VA" device named M1, and THAT device (YPSP103_VA!M1) is the one that has a GM variable to print.   To print its GM, one would have to print N(XM1:YPSP103_VA!M1:GM). 

In short, whenever you have devices inside subcircuits and want to print information specific to those low-level devices (such as output variables) you have to specify a "path" through all subcircuit layers all the way down to the actual low-level device of interest.

For really complex PDKs with deep nesting of subcircuits this can be hard.  For this PDK I think it is relatively easy.

It would be nice if Xyce's `.op` actually printed all the output variables.  But that is a very longstanding request in the Xyce product that has been unaddressed for many years.

tvrusso, 2025-01-14:
As for the claim that Xyce's reference guide contains no list of output variables for the PSP 103 (level 103 MOSFET model), there is a table of them in the 7.8 reference guide beginning on page 519.

It's even there in the 7.6 reference guide starting on page 518.

tvrusso, 2025-01-14:
A little more information here (it is difficult to stop me once I get started):

If you look at the output of `.op` above, you'll see that it is divided up into multiple sections --- one section is outputting the model parameters from `.model` cards, and the other is outputting instance parameters for actual devices.  So, for example:

```
Number of ADMS PSP103_VA instances: 2
          name XM1:YPSP103_VA!M1XM2:YPSP103_VA!M1
         model XM1:SG13G2_LV_NMOS_PSPXM2:SG13G2_LV_PMOS_PSP
```

Here's where you actually find the real names of the mosfet devices, such as `XM1:YPSP103_VA!M1` and `XM2:YPSP103_VA!M1` (the lack of space between the two is another longstanding bug in `.op` output that has never been addressed --- the columns are fixed width and the column headers run together if the names are too long).

These are the names you actually need to use in N() syntax to get at the output variables.

Final comment:  Like Adrien, I have no access to the slack channel.  I'm a retiree from the Xyce team and this means that I don't get access to that channel, but it also means I've got a *lot* more time than the overworked current members of the Xyce team to answer questions like this and others.   The Github issues and discussions sections for the Xyce project and the Xyce google mailing list are the only ways I have to do that.

KrzysztofHerman, 2025-01-15:
Thank you @tvrusso  for the detailed description. @adrienluitot were you able to get the value of `gm` using the presented method ?

adrienluitot, 2025-01-15:
> In short, whenever you have devices inside subcircuits and want to print information specific to those low-level devices (such as output variables) you have to specify a "path" through all subcircuit layers all the way down to the actual low-level device of interest.

That's what I thought, like in Ngspice, but I didn't know how the path worked. And also the error "Function or variable N(MN0:GM) is not defined" was not clear to me, I wasn't sure it meant a wrong component name/path.
‎.
 
> As for the claim that Xyce's reference guide contains no list of output variables for the PSP 103 (level 103 MOSFET model), there is a table of them in the 7.8 reference guide beginning on page 519.

My bad, I saw level 103 MOSFET output variables indeed, but I wasn't sure it was the same as PSP103.
.

 >  @adrienluitot were you able to get the value of gm using the presented method ?
 
Yes! I used the syntax given by `.op` as @tvrusso said. Thanks a lot  for your help!

KrzysztofHerman, 2025-01-15:
ok, so I will close as solved. Once again many thanks to @tvrusso

closed 2025-01-15 by KrzysztofHerman

adrienluitot, 2025-01-15:
Just as an example, I ran:

```spice
.SUBCKT IHP_PDK_nonlinear_components_sg13_lv_nmos  gnd d g s b w=0.35u l=0.34u ng=1 m=1 as=0 ad=0 pd=0 ps=0 trise=0 z1=0.346e-6 z2=0.38e-6 wmin=0.15e-6 rfmode=0 pre_layout=1 mlist=1
X1 d g s b  sg13_lv_nmos w={w} l={l} ng={ng} m={m} as={as} ad={ad} pd={pd} 
+ ps={ps} trise={trise} z1={z1} z2={z2} wmin={wmin} rfmode={rfmode} pre_layout={pre_layout} 
.ENDS
  
.lib /home/user/Documents/silicon/IHP-Open-PDK/ihp-sg13g2/libs.tech/xyce/models/cornerMOSlv.lib mos_tt

* Gate voltage source
V1 input 0 DC 1
* Drain voltage source
V2 _net0 0 DC 3.3

XMN0 0  _net0 input 0 gnd IHP_PDK_nonlinear_components_sg13_lv_nmos w=0.35U l=0.34U ng=1 m=1 as=0 ad=0 pd=0 ps=0 trise=0 z1=0.346E-6 z2=0.38E-6 wmin=0.15E-6 rfmode=0 pre_layout=1

* Sweep gate's voltage between 1 and 2 volts
.dc LIN V1 1 2 0.5

.PRINT dc format=csv file=output.csv V(input) N(XMN0:X1:YPSP103_VA!M1:gm)

.END
```

Which gave the output:
```csv
V(INPUT),	N(XMN0:X1:YPSP103_VA!M1:GM)
1.00000000e+00,	1.99032378e-04
1.50000000e+00,	1.95248414e-04
2.00000000e+00,	1.78676974e-04
```