IHP-GmbH/IHP-Open-PDK #330: Cannot retrieve lv_nmos model parameters (e.g. gds, cgs)

arnaudgabas, 2025-01-14:
Hello,

I have made a simple circuit to try to retrieve the lv_nmos parameters such as gm, gds. When simulating with ngspice and Qucs-S, I get this : 

[image]

The gm value is plotted with the sweep, but the other only gives results on the first step. I have found the mos parameters in the PSP103_module.include file in the PDK.

Does anyone have an idea of what could be the problem?

KrzysztofHerman, 2025-01-14:
@arnaudgabas could you please share your testcase (schematic and netlist) ?

arnaudgabas, 2025-01-14:
Thank you for your fast answer. Here it is : 

[testcase.tar.gz](https://github.com/user-attachments/files/18407862/testcase.tar.gz)

arnaudgabas, 2025-01-16:
Hello @KrzysztofHerman,

Do you have any news about it ? I have also found that adding a "DC Simulation" block in the schematic breaks everything. It seems that I am not using Qucs correctly for this.

KrzysztofHerman, 2025-01-16:
@arnaudgabas 
I think I have just hit the solution 

[image]

As you can see in the `INCLUDE SCRIPT` I have changed `.SAVE` to `save all`. Now it works

arnaudgabas, 2025-01-16:
Thank you for your answer.

I tried to edit it on my circuit but nothing changed. 

Can you please send me the schematic ? Also, what are your current Qucs/ngspice versions ? We have noticed that depending on the OpenVAF/Qucs/ngspice versions, some features doesn't work as expected.

KrzysztofHerman, 2025-01-16:
@arnaudgabas I use `qucs-s 24.4.1` and `ngspice 44`

[qucs-s_schematic.sch.gz](https://github.com/user-attachments/files/18436694/qucs-s_schematic.sch.gz)

KrzysztofHerman, 2025-01-16:
@arnaudgabas also please take a look at the table configuration and the way you source the data

[image]

arnaudgabas, 2025-01-16:
I found the problem : it was due to a Qucs-S bug that does not refresh the content of the `include script` correctly.

I fixed the bug by deleting the script in my schematic and rewriting it with `.save all`. Thank you for your help!

closed 2025-01-16 by arnaudgabas