IHP-GmbH/IHP-Open-PDK #332: pTap guardring LVS issues

PhillipRambo, 2025-01-15:
Problem Description: pTap Guard Ring LVS Issue

In the test case, a pTap guard ring was created with an NMOS placed inside. During LVS:

    All NMOS terminals are detected as connected to the guard ring.

Possible cause: The TEXT.drawing layer with the sub! annotation might be introducing the error. The structure was copied from PDK LVS examples, so there might be an issue with how it's implemented. Below the testcase is attached and can be reviewed. 

By the way, im not experiencing these issues with nTap generation. 
[Uploading test_case(1).zip…]()

PhillipRambo, 2025-01-15:
[test_case(1).zip](https://github.com/user-attachments/files/18424400/test_case.1.zip)

sergeiandreyev, 2025-01-17:
Hi @atorkmabrains, @FaragElsayed2, could you please take a look at this issue?

atorkmabrains, 2025-01-20:
Hi @PhillipRambo and @sergeiandreyev 

We will get to this soon.

FaragElsayed2, 2025-01-21:
Hi @PhillipRambo 

The ntap1 and ptap1 devices will be fully updated soon to align with the internal discussions.

FaragElsayed2, 2025-01-21:
Hi @PhillipRambo @sergeiandreyev 

As discussed in Slack, LVS for ptap1 and ntap1 devices will be updated soon to align with the commercial PDK.

FaragElsayed2, 2025-01-29:
Hi @PhillipRambo 

LVS has been updated for the n/p tap1 devices.

Here is the updated CDL that passes LVS. The shared version does not match the layout in the test case.

```
.subckt input_stage vss vdd dn4 dn3 dn2 iout
*.PININFO vss:B vdd:B dn4:B dn3:B dn2:B iout:B
M7 dn2 iout vdd vss sg13_lv_nmos l=0.13u w=0.15u ng=1 m=1
R2 tie vss ptap1 A = 7.6842p P=56.92u
M1 dn3 dn3 vss vss sg13_lv_nmos l=9.75u w=720n ng=1 m=1
M2 dn4 dn3 vss vss sg13_lv_nmos l=9.75u w=720n ng=1 m=1
.ends
```

sergeiandreyev, 2025-03-17:
Hi @PhillipRambo, could you please double check and confirm if the issue fixed?

closed 2025-05-07 by PhillipRambo