IHP-GmbH/IHP-Open-PDK #334: NoMet4Filler and NoMet5Filler missing in inductors_code.py

VolkerMuehlhaus, 2025-01-16:
Metal4.nofill and Metal5.nofill is missing for inductor2 and inductor3 library elements

```
        dbCreatePolygon(self, PWellBlock, polyPoints1)
        dbCreatePolygon(self, NoActFiller, polyPoints1)
        dbCreatePolygon(self, NoGatFiller, polyPoints1)
        dbCreatePolygon(self, NoMet1Filler, polyPoints1)
        dbCreatePolygon(self, NoMet2Filler, polyPoints1)
        dbCreatePolygon(self, NoMet3Filler, polyPoints1)

        dbCreatePolygon(self, NoTMet1Filler, polyPoints1)
        dbCreatePolygon(self, NoTMet2Filler, polyPoints1)
```

sergeiandreyev, 2025-01-21:
Hi @VolkerMuehlhaus, could you please confirm that the issue is fixed from your side as well, so that we can close the ticket?
The updates are in the latest `dev` branch.

VolkerMuehlhaus, 2025-01-21:
> Hi [@VolkerMuehlhaus](https://github.com/VolkerMuehlhaus), could you please confirm that the issue is fixed from your side as well, so that we can close the ticket? The updates are in the latest `dev` branch.

Hi Sergei, looks good to me!

closed 2025-01-21 by sergeiandreyev