IHP-GmbH/IHP-Open-PDK #337: Can't Solve Fill Cell/Density DRC Error

shafinhossain, 2025-01-21:
I initially ran the DRC without using any fill cells and I encountered multiple density errors.

[image]

Then I ran the scripts for active, metal and top metal fillers and even ran the density report.
After that I executed the DRC again and encountered the exact density DRC errors before using the filler cells. Could anyone kindly walk me through this issue?

[image]

Note: I did not draw any PR boundaries, could that be a factor? and if I had to draw one how would I approach?

PhillipRambo, 2025-01-21:
Hi @shafinhossain could you please provide the testcases so i can reproduce the issues. Are you working on the latest commit?

Kindes Regards Phillip

shafinhossain, 2025-01-21:
Hello @PhillipRambo 
Here you can find the testcase, file name is 'test_case_cell':
https://drive.google.com/drive/folders/1FOBsojHR_wbLBcXiUR6d9uIEcFulxXRO?usp=sharing

Ikram-rs22, 2025-01-22:
@PhillipRambo any update on this issue?? Facing the same problem.

akrinke, 2025-01-22:
I think this problem is caused by the chip being too small. Since the area at the border is not filled, the overall density becomes too low. Maybe you can increase the fill density in the scripts?

Ikram-rs22, 2025-01-22:
Do I reduce it? I've already reduced reduced the spacing of active filler to 0.42 which I assume to be the drc limit. 

[image]

Does the other filler layers (gatpoly, metal, top metal) have the same drc limit of 0.42?

akrinke, 2025-01-22:
I used a value of 0.5 and it removed most of the DRC errors regarding active density. The remaining 3 errors are due to the 800x800 µm² window that is used for density calculation. I think they can be ignored.

Ikram-rs22, 2025-01-22:
Yes I can confirm that after resizing the seal ring to 750x750 and using spacing for both active & gate poly filler to 0.42, the drc reduced from 33 to 17

PhillipRambo, 2025-01-22:
And what are the remaning issues related to then? Are you using the testcase provided by @shafinhossain?

Ikram-rs22, 2025-01-22:
No I'm using a separate testcase. I don't know about @shafinhossain testcase

shafinhossain, 2025-01-22:
@PhillipRambo I ran the DRC using 750x750 um seal ring, with 0.5 filler script spacing.

[image]

I got 21 errors in the first run, in the second run oddly I got 17.

[image]

shafinhossain, 2025-01-23:
@PhillipRambo did you follow up on this issue? it would be really helpful for me. 
TIA.

sergeiandreyev, 2025-01-23:
Hi @shafinhossain, could you please make a screenshot with the DRC violations expanded or send us the `.rdb` file

shafinhossain, 2025-01-24:
Hello @PhillipRambo @sergeiandreyev 
I am using the latest commit and dev branch, I am filling all the layers and I do not know how to exclude any.

[image]
I filled activ/gatpoly, metal and topmetal and then ran a density report.

Here's the snap of the stretched error box.

[image]

I am also dropping the .rdb file link here.
https://drive.google.com/file/d/1oSMQ-p9HM8hUVaG0fkZtmimucAjOG7Pv/view?usp=sharing

If you need anything else please do let me know. Really grateful for you responses.

shafinhossain, 2025-01-27:
any update on the matter @PhillipRambo @sergeiandreyev ?
TIA. :D

PhillipRambo, 2025-01-27:
Hi @shafinhossain,

After reviewing the testcase, we’ve identified that the filler script needs optimization to address the density errors and related issues.

If your goal is to submit the design for an OS tapeout, a possible workaround would be to send the final design to us, and we’ll take care of adding the necessary fillers for you.

@dnltz, could you please take a look at the testcase? It serves as a good example of how the filler script could be improved for small designs like this.

Best regards,
Phillip

shafinhossain, 2025-01-28:
Thanks a lot @PhillipRambo. I do want to submit my design for the March Tapeout. If I clean the rest of the drcs (ones except the density and filler cells), could you be kind enough to take care the rest of it as you mentioned?

Regards,
Shafin Hossain

PhillipRambo, 2025-01-30:
@dnltz Hi Daniel. Thanks for the update, we tried playing around with some different spacing settings and i can then meet the filler rules for the design given in this example, but you are right a more sofisticated approach to filling is needed. Thanks for the help!

closed 2026-01-15 by sergeiandreyev