IHP-GmbH/IHP-Open-PDK #340: Need the Noise Test Circuits of HBTs

EngGhaith, 2025-01-21:
Hi!
I am trying to simulate transient noise in XYCE.
I have created 2 models of NPN-transient noise and need to verify them.
The parameters (RE, RSUB, ...etc.) are taken from VBIC-Model and their values can be read from .OP simulation.
Thermal noise and shot noise are white, Flicker noise is 1/f-distributed.
The values of Voltage/Current sources can be loaded from a file.

[image]

I have seen that IHP has done a noise figure measurements and the results are available in the IHP-Open-PDK repository, but these measurements are not supplied with test circuits (Input-Terminal? Output-Terminal? Supply Voltage? Matching circuits to know noise bandwidth? ...etc.).
Could you please provide this circuits to verify my models?

Thak you in advance.
AL SABAGH Ghaith from IIC-JKU.

sergeiandreyev, 2025-01-22:
Hi @EngGhaith, thank you for the report! I'll check internally if we can find / provide test circuits..

sergeiandreyev, 2025-01-23:
@EngGhaith, could you please clarify which are the exact files that you were reviewing in our repo for the noise measurement data?

EngGhaith, 2025-01-23:
> [@EngGhaith](https://github.com/EngGhaith), could you please clarify which are the exact files that you were reviewing in our repo for the noise measurement data?

ihp-sg13g2/libs.doc/meas/HBT/doc/html_npn13g2l_VBIC/High_Frequency_Noise_SG13G2_pdk.pdf
ihp-sg13g2/libs.doc/meas/HBT/doc/html_npn13g2l_VBIC/Low_Frequency_Noise_SG13G2_pdk.pdf

xcub, 2025-01-23:
Dear @EngGhaith,
As I can see from the noise measurement report, it was a common-emitter (input: base, output: collector, ground: emitter) configuration. The operating points (Vbe and Vce, frequencies, emitter lengths, and fingers) are also mentioned in the report. Please note that no matching network was used in these measurements.

I hope this information helps you.

WIth regards
Farabi

FarabiJamal, 2025-01-23:
Dear @EngGhaith,
As I can see from the noise measurement report, it was a common-emitter (input: base, output: collector, ground: emitter) configuration. The operating points (Vbe and Vce, frequencies, emitter lengths, and fingers) are also mentioned in the report. Please note that no matching network was used in these measurements.

I hope this information helps you.

WIth regards
Farabi

FarabiJamal, 2025-01-29:
> > Dear [@EngGhaith](https://github.com/EngGhaith), As I can see from the noise measurement report, it was a common-emitter (input: base, output: collector, ground: emitter) configuration. The operating points (Vbe and Vce, frequencies, emitter lengths, and fingers) are also mentioned in the report. Please note that no matching network was used in these measurements.
> > I hope this information helps you.
> > WIth regards Farabi
> 
> Dear [@FarabiJamal](https://github.com/FarabiJamal) Do you know which VCC, RB, RE and RC do we have (if we have any resistors)? And which Input amplitudes were used?

Dear @EngGhaith,
The Vcc would be Vcb+Vbe from the reports. There was no RB, RE and RC. The attached document can give you better insight about noise measurements and modeling.

With regards
Farabi

[NOISEdoc.pdf](https://github.com/user-attachments/files/18587548/NOISEdoc.pdf)

sergeiandreyev, 2025-01-31:
Also @EngGhaith, could you please attach actual testcase schematic netlist file to this GitHub issue?

EngGhaith, 2025-02-02:
@sergeiandreyev 
Yes I can
Here are 3 files:
- "create_noise_NPN_TRNOISE_TB.py" would create the required noise samples. You to run this firstly!
- Then use the schematic: "NPN_TRNOISE_TB.sch"
- At the end plot the results with: "plotFFTNPN_TB.py"

PS.: Here I have neglegated the Flicker noise. Because in such a frequency (12 GHz) it would be very small compared to shot/thermal.
If I would modeled that, then it would require a long simulation time (e.g. 1s to achieve one period of the worst frequencies in lower frequency band) and a fine teime step (e.g. tstep=40ps) to get enough resolution for the tone 12 GHz). That would require about 25 GSample for each vector, which would be K.O. for my laptop.

You will see that the model would make different values if you change the input amplitude, thus a proper explaination of IHP-used-setup would be necessary to compare results!

[noise_issue.zip](https://github.com/user-attachments/files/18633132/noise_issue.zip)

EngGhaith, 2025-02-02:
> [@sergeiandreyev](https://github.com/sergeiandreyev) Yes I can Here are 3 files:
>
> * "create_noise_NPN_TRNOISE_TB.py" would create the required noise samples. You to run this firstly!
> * Then use the [...]

If you wanna activate the flicker noise, just enable their elements in the schematics with shift+t after selection, and uncomment the dedicated lines in python after modifying them