IHP-GmbH/IHP-Open-PDK #346: DRM - pSD section - ThickGateOx mentionned but not shown

adrienluitot, 2025-01-29:
Hello, I don't know if it's a mistake an omission but in the `pSD` part (p 28 in rev 0.4) of the DRM, the `ThickGateOx` is mentioned in the table (first picture) but never shown in the drawing (second picture).

**PS:** I actually realised that the `ThickGateOx` is shown, but not captioned. `ThickGateOx` and `pSD` basically have the same color... a bit tricky...
Maybe the color of one of them should be changed in the whole doc to avoid this kind of omission?

- - -
[image]
- - -
[image]
- - -

sergeiandreyev, 2025-01-30:
Hi @stefansimon42, could you please take a look?

adrienluitot, 2025-01-30:
I take this opportunity to report another little mistake. Rule `MIM.i` was removed from the in revision 0.2 (I supposed from the changelog) but is still present on the diagram.

[image]

stefansimon42, 2025-02-06:
Hi @adrienluitot,

Thanks for the hint regarding Mim.i

W.r.t. the PSD rules, only the PSD layers are actually drawn in the illustration. You are right that there is a risk of confusion between pSD and ThickGateOx. The layer colors are adapted to the representations in the layout editors to make them easier to understand. We will discuss an adjustment of the colors internally.

However, I am of the opinion that pSD.i1 and pSD.j1 do not necessarily have to be drawn in this figure. The amount of information in relation to pSD.i and pSD.j is marginal, but the drawing itself would be cluttered. I tend not to add the drawings for pSD.i1 and pSD.j1.

adrienluitot, 2025-02-06:
I agree that that showing the thick oxide for the rules i1 and j1 would overwhelm the manual. It’s all my fault and I thought psd.i and psd.j were related to thick oxide and that it was shown on the drawing but not in the legend. 
I think I mainly misunderstood/misread, and a little due the color I guess.

Maybe the “NOT in Thick Oxyde” is a bit confusing, I mean the “not” is easy to miss.
I understand it’s here because of the contradiction with i1 or j1. 
So maybe just a sentence like “min pSD space to NFET Gate” would be enough and reduce confusion?
Just an idea, do what you want with it.