IHP-GmbH/IHP-Open-PDK #347: LVS behaviour with parallel MIM Capacitors

adrienluitot, 2025-01-30:
When I stick 2 MIM caps with through Metal 5 and TopMetal 1 to use them in parallel, the LVS won't detect the capacitors at all. (Case 1)
If I space them, the LVS will detect 2 independent MIM caps. (Case 2)
And if keep the space, but fill it with Metal 5 and TopMetal 1 to connect them as if they were stuck, the LVS detect 1 MIM cap with a multiplier=2. (Case 3)

For the case 2 and 3, the space can be as small as 0.005um.

In the end, what I would expect, is a Layout like in case 1, but with LVS results like in case 3.
I don't really understand why the LVS results between case 1 and 3 are so different.

Also, maybe I'm not doing the parallelisation properly?

---
### Case 1
Layout:
[image]
LVS: <empty>

---
### Case 2
Layout:
[image]
LVS:
[image]

---
### Case 3
Layout:
[image]
LVS:

[image]

sergeiandreyev, 2025-02-04:
Hi @atorkmabrains, @FaragElsayed2, could you please take a look?

FaragElsayed2, 2025-02-04:
Hi @adrienluitot 

Could you share the test cases you mentioned?

adrienluitot, 2025-02-10:
Hello @FaragElsayed2, sorry for the late reply...

Here, [a zip containing a GDS](https://github.com/user-attachments/files/18739403/test_capa_lvs.gds.zip) with the 3 cases and a bit of text for each.
I won't share a netlist because it doesn't matter for the LVS, you can use any netlist I think.

Thank you for your interest!

adrienluitot, 2025-03-29:
Anything new @FaragElsayed2 ?

FaragElsayed2, 2025-04-03:
Hi @adrienluitot 

Sorry for the delay, I missed your reply. Thanks for sharing this testcase.

I've updated the current MIM derivations to support all these cases. Please recheck and let me know if you still encounter this issue.

FaragElsayed2, 2025-08-28:
@sergeiandreyev I believe this issue could be closed.

atorkmabrains, 2026-01-15:
@sergeiandreyev and @KrzysztofHerman Could you please close this issue?