IHP-GmbH/IHP-Open-PDK #349: Hybrid EM-Circuit Simulation Workflow for IHP SG13G2

rupokdas147, 2025-02-03:
I am designing a 100 GHz broadband power amplifier using the IHP SG13G2 HBT process. I can individually EMS transmission lines and inductors using openEMS and use the s2p files in the schematic. But this method does not include the effects of nearby signal lines, since I am EMS them individually. But I want to validate my hybrid EM-circuit simulation more accurately. But openEMS does not support active devices like HBT.  So, I plan to:

1. Define multiple ports at all transistor terminals (base, collector, emitter) and interconnect points in openEMS.

2. Simulate all passive structures together (transmission lines, inductors, matching networks) to capture mutual coupling and parasitic effects.

3. Extract an SNP file (e.g., S4P, S6P) containing the S-parameters of the full passive layout.

4. Import the SNP file into Qucs-S and connect:
BJTs using IHP's SG13G2 models, Bias sources at the correct terminals, Additional lumped elements as needed.

5. Run circuit simulations to analyze gain, stability, and nonlinear performance.

Questions:
Is this approach correct for hybrid EM-circuit simulation with IHP's SG13G2 process?
Should I terminate every open terminal with a port in openEMS?
Are there better methods for integrating EM-simulated layouts with IHP’s active device models?

Any insights or corrections would be greatly appreciated!

Thank you!

VolkerMuehlhaus, 2025-02-04:
@rupokdas147 Yes, this is what I meant. 

But to avoid misunderstanding: a pair of differential lines should not be split, of course. I only suggest to simulate those parts separately where coupling is not desirable and not expected.

rupokdas147, 2025-02-05:
@VolkerMuehlhaus If I create a PEC layer below the Substrate to use it as a global GND, will it work? Does the length of the port matter?

VolkerMuehlhaus, 2025-02-05:
> [@VolkerMuehlhaus](https://github.com/VolkerMuehlhaus) If I create a PEC layer below the Substrate to use it as a global GND, will it work? Does the length of the port matter?

Yes, the length of the port matters, it behaves like a physical wire. Long ports create parasitic inductance (from length) and couple to other layout parts due to current flowing in the port. 

For openEMS and other 3D EM tools like HFSS, the lumped port must be short. Having a port from TopMetal2 down to Metal1 is ok. Having a port across some 100µm down to the backside of the substrate will not give useful results. 

In the link below, you find the screenshot of 2-port simulation model of an octagon inductor, where we have no nearby ground in layout. I added an artificial lossless metal polygon at the top surface of the substrate, and used that as common ground ref for both ports. By placing this extra metal on the surface of the substrate, shunt path from inductor to substrate is included in the results. I made this extra metal lossless because the port return current will physical flow through that metal strip -> don’t add resistance here. It add some small inductance, but that’s nothing to worry about, very few pH.

https://muehlhaus.com/wp-content/uploads/2025/02/inductor_twoport_extrametal.png

rupokdas147, 2025-02-05:
@VolkerMuehlhaus Let's say, I want to EMS a half circle inductor. How can I do that in openEMS? To connect the GND of two ports, I have to use a long lossless metal below or above the TopMetal through the diameter of the circle, If I want a short path. But this will add more inductance to the result.

VolkerMuehlhaus, 2025-02-05:
> [@VolkerMuehlhaus](https://github.com/VolkerMuehlhaus) Let's say, I want to EMS a half circle inductor. How can I do that in openEMS? To connect the GND of two ports, I have to use a long lossless [...]

Correct, for all volume meshing 3D EM tools you need to "close the loop" in some way. So you can't simulate that half circle without adding some return to close loop. 

One approach used by the Sonnet solver (MoM, but with similar requirement of "physical" ports) is to use a PEC simulation boundary as a global return path (i.e. device in a metal box), with some length of line added between the box wall port and the device under test. That length of line must then be removed by port calibration, and the remaining results is the half circle. 
We can't use the exact same port calibration, but the general idea of adding some metal to close the loop and removing the effect of that metal by de-embedding would be applicable. Similar to removing the pads in on-wafer measurements. There is no such de-embedding implemented in openEMS, but scikit-rf might be useful here for postprocessing.

And in real life, your current also flows in closed loops, so you might include the true return path layout in your model, instead of cutting out only the half circle for EM. There will be some influence by such a metal frame on inductance anyway, so why not include that in the EM model from the very beginning?

I know that this is unexpected for ADS Momentum users who can use an abstract "ground at infinity" or ground at the bottom of the substrate, with current jumping in space between different locations with no inductance or delay. But it is what it is ... the model in 3D volume mesh EM requires the same as real world physics: current flows in closed loops only. We need to consider this for modelling, and can't cut out fractions for EM.

VolkerMuehlhaus, 2025-02-05:
@rupokdas147 Thanks for your input! I am currently working on a similar use case for a colleague at IHP, to build models and see what port configuration works best. The plan ist to create a document with more detailed instructions and examples. Let's see what we get ...

rupokdas147, 2025-02-06:
@VolkerMuehlhaus What's the best way to handle vias in EM simulation? A lot of via slow down the speed of the simulation significantly.

rupokdas147, 2025-02-06:
> > [@VolkerMuehlhaus](https://github.com/VolkerMuehlhaus) What's the best way to handle vias in EM simulation? A lot of via slow down the speed of the simulation significantly.
> 
> The "extra cost" of simulating many vias in FDTD is not as bad as in MoM, because the matrix size will grow only linear with number of mesh cells, but in many cases you want to merge via arrays before simulation. This is not yet implemented in the old GDSII import code [here](https://github.com/IHP-GmbH/IHP-Open-PDK/tree/main/ihp-sg13g2/libs.tech/openems/import_GDSII).
> 
> I am currently working on a new Python workflow which simplifies usage a lot, and enables via merging. This is not released yet, but in internal testing now.
> 
> Do you use openEMS with Python or Matlab code?
> 

I'm using python for simulation. Maybe simulation time is not affected, the 3D view was affected.

rupokdas147, 2025-02-07:
@VolkerMuehlhaus Manually meshing is a problem. Taking port dimension less than mesh size, does not start the simulation (Energy remains 0 forever).

VolkerMuehlhaus, 2025-02-07:
@rupokdas147 Absolutely correct! If you are interested to beta test my new SG13 openEMS Python workflow including automatic, please contact me at volker@muehlhaus.com  
It is not on Github yet because I am still fine tuning some details, but you can already give it a try, if you wish.

gvidasgsasicnl, 2025-02-07:
[like]  Gvidas Sidlauskas reacted to your message:

rupokdas147, 2025-02-09:
> @rupokdas147 Absolutely correct! If you are interested to beta test my new SG13 openEMS Python workflow including automatic meshing, please contact me at volker@muehlhaus.com  
> It is not on Github yet because I am still fine tuning some details, but you can already give it a try, if you wish.

Thanks, I have mailed you from rupokdas147@gmail.com