IHP-GmbH/IHP-Open-PDK #350: m parameter in lv_mos models of Xyce in the current dev-branch dont work

olisnr, 2025-02-06:
to demonstrate it, i added two transistors to the `ihp-sg13g2/libs.tech/xschem/sg13g2_tests_xyce/ac_lv_nmosrf.sch`. one with m=10 and the other with w=10u (standard is 1u).

i would expect similar results, but the m=10 device looks exactly like the m=1 device with the same w...

[ac_lv_nmosrf_mult_Xyce.sch.zip](https://github.com/user-attachments/files/18690965/ac_lv_nmosrf_mult_Xyce.sch.zip)

[image]

dwarning, 2025-02-09:
I can confirm that multiplier didn't work with Xyce. It seems that Xyce can not change parameter "m" in subckt call.
Changing this parameter name to "mult" or something else will work as in following script:

`psp103 nch transfer

vd  d 0 dc 0.1
vg  g 0 dc 0.0
vb  b 0 dc 0.0
XPSP103_VA d g 0 b psp_sub w=10u l=1u mult=2

.subckt psp_sub d g s b w=10u l=1u mult=1
YPSP103_VA m1  d g s b nch l='l' w='w' mult='mult'
.ends

.option temp=21
.step  vb -3 0 0.5
.dc vg 0 1.5 0.02
.print dc format=gnuplot v(g) i(vd)

.model nch psp103_va type=1

.end
`

dwarning, 2025-02-12:
Xyce team has agreed that the parser has a [bug](https://github.com/Xyce/Xyce/issues/134).
Change parameter name to mult instead m for symbols and subckt definitions for xyce and ngspice lib's is preferable. Positive side effect is that the meaning for the user is unique.

tskaar, 2025-07-30:
For your reference, it's fixed in Xyce v7.10:

Xyce v7.10:
---

[image]

NGSpice
---

[image]

Using the same testbench as opening post, just modified it such that left plot is always Xyce and right plot is always NGSpice

olisnr, 2025-08-20:
@sergeiandreyev check

closed 2025-08-20 by olisnr