IHP-GmbH/IHP-Open-PDK #351: Why is "ng" a divisor? And one more question...

MilosMitrovic411, 2025-02-08:
There are already two issues that are kind of tangential to my question (https://github.com/IHP-GmbH/IHP-Open-PDK/issues/203 and https://github.com/IHP-GmbH/IHP-Open-PDK/issues/180). I've voiced my opinion on both these issues, but I think it is worth of having it's own topic.

In my experience, PDK vendors use "ng" (or "nf") as a multiplier. e.g. if you want a transistor that has a total width of 10u and has 4 gates (fingers), you would annotate it as: w=2.5u, m=1, ng=4.

It would be very confusing to me if I would have four 2.5u fingers and a property that says that my instance is 10u wide. I don't know what are the parameters of the transistors generated form the schematics in KLayout, or if there is even that option.

I am not using  software recommended in the official PDK, actually I am working on my own version of SG13G2 PDK for Glade (which is similar to Virtuoso - allows both design and layout in one software), and I am using "ng" as a multiplier, that's why I was surprised when I saw in nmos model code "ng" being used as a divisor.

On that note, I have read the Apache License 2.0, but I would like to confirm with somebody employed with IHP, is it within the law for me to publish my own version of this PDK? I am also open for my files being included in this repository.

dwarning, 2025-02-09:
Which PDK use finger count as multiplier?

All actual MOS models I know using W as total width and finger width is W/NF. So we have joint areas for drain and/or source which are calculated separately.

A multiplier describes total separate devices in parallel and should not confused with finger count.

There is no bug here and issue should be closed.

MilosMitrovic411, 2025-02-10:
From my experience, TSMC and NXP use it as such, can't say which specific technologies because of NDAs.
As I said in the post it is from my experience, I heard from other designers that they encountered both representations.

This was not meant as an attack, my question is aimed towards understanding the logic behind it given that during a layout extraction each finger is extracted as a separate transistor. So in my interpretation one starts from there and works towards annotation on the schematic.

I didn't say it is a bug, although the issue https://github.com/IHP-GmbH/IHP-Open-PDK/issues/203 is directly caused by that approach. But, hey, I guess you are right, this is more of a discussion topic not an issue, so I will post it there.

closed 2025-02-10 by MilosMitrovic411